meta-qt5/meta-qt5 #86: recipes-qt/qt5/qt3d_git.bb fails to compile

mortie, 2018-08-01:
I'm trying to cross compile some Qt code for aarch64, following https://developer.toradex.com/knowledge-base/how-to-set-up-qt-creator-to-cross-compile-for-embedded-linux. `bitbake -k meta-toolchain-qt5` fails  when trying to compile qt3d. The complete log file is attached, but the relevant part is here:

```
make[3]: Entering directory '/data/martin/yocto/build/tmp/work/aarch64-poky-linux/qt3d/5.11.1+gitAUTOINC+3b4a9b5061-r0/build/tests/manual/gltf'
make[3]: Nothing to be done for 'first'.
make[3]: Leaving directory '/data/martin/yocto/build/tmp/work/aarch64-poky-linux/qt3d/5.11.1+gitAUTOINC+3b4a9b5061-r0/build/tests/manual/gltf'
No data signature found
Makefile:460: recipe for target '.rcc/qrc_scene2d.o' failed
make[3]: *** [.rcc/qrc_scene2d.o] Error 1
make[3]: Leaving directory '/data/martin/yocto/build/tmp/work/aarch64-poky-linux/qt3d/5.11.1+gitAUTOINC+3b4a9b5061-r0/build/examples/qt3d/scene2d'
Makefile:490: recipe for target 'sub-scene2d-make_first' failed
make[2]: *** [sub-scene2d-make_first] Error 2
make[2]: Leaving directory '/data/martin/yocto/build/tmp/work/aarch64-poky-linux/qt3d/5.11.1+gitAUTOINC+3b4a9b5061-r0/build/examples/qt3d'
Makefile:44: recipe for target 'sub-qt3d-make_first' failed
make[1]: *** [sub-qt3d-make_first] Error 2
make[1]: Leaving directory '/data/martin/yocto/build/tmp/work/aarch64-poky-linux/qt3d/5.11.1+gitAUTOINC+3b4a9b5061-r0/build/examples'
Makefile:98: recipe for target 'sub-examples-make_first' failed
make: *** [sub-examples-make_first] Error 2
make: *** Waiting for unfinished jobs....
```

[log.do_compile.11434.txt](https://github.com/meta-qt5/meta-qt5/files/2248868/log.do_compile.11434.txt)

gvalcaza, 2018-08-31:
This is also happening to me quite often, on a different file but with the same rcc error (No data signature found)

Meisterdaxs, 2018-09-18:
Same issue here. Could that be related to kind of concurrent builds? It happens from time to time only.

denix0, 2018-10-17:
Well, me too... :(

gvalcaza, 2018-12-28:
This bug has forced my development team to remove qt3d altogether from our Yocto builds to avoid the risk of our daily builds failing randomly. Is there any more information on this bug and/or a possible fix? I've been trying to debug it on my own to no avail...

SamBissig, 2019-01-08:
We also still struggling with the issue. To further understand the issue some questions to see if you have the same setup as we have:

- Do you build one image at time or are you running multiple builds accessing the sam SSTATE cache the same time? Could it be a race contidtion on the SSTATE cache?
- Do you guys all build the Toradex BSP?

platisd, 2019-02-14:
I'm also facing this issue trying to build our Yocto (Rocko 2.4.2) image for Raspberry Pi 3. The main difference with the setup we have our CI machines is that my host is running **XUbuntu 18.04 LTS** so, I wonder, does that shed any light on the probable culprit?

justdoGIT, 2019-04-04:
I am also facing this issue after upgrading the meson and the libinput library. I don't know it has any relation with that or not. But before that, it was compiled successfully.

martin-ejdestig, 2019-04-19:
Have had random compilation failures of rcc generated files in qt3d as well. Caused by GCC OOM:ing (easy to see with journalctl).

Only way to solve this for my particular setup was to add a 64GB swap file. (Depends on available memory and #of parallel jobs, but this was the only thing that helped in this case. Compiling just a single failing compilation unit and nothing else at the same time still failed.)

Is there a way to disable examples (and tests would be nice as well) globally in meta-qt5? Would make builds go faster and this particular issue would go away. As far as I can tell only qtbase has a PACKAGECONFIG for disabling examples:

https://github.com/meta-qt5/meta-qt5/blob/a4bf2402c19742b866cf4b5e5e4665ae070500ff/recipes-qt/qt5/qtbase_git.bb#L104

Could add that PACKAGECONFIG to qt3d as well but would be nice to be able to do it globally. Am I missing something?

churmenton, 2020-01-15:
@martin-ejdestig 
meta-qt5/recipes-qt/qt5/qt5.inc
add 
PACKAGECONFIG_remove = "examples"
before
EXTRA_QMAKEVARS_PRE += "${@bb.utils.contains('PACKAGECONFIG', 'examples', 'QT_BUILD_PARTS+=examples', '', d)}"

churmenton, 2020-01-15:
@martin-ejdestig 
meta-qt5/recipes-qt/qt5/qt5.inc
add
PACKAGECONFIG_remove = "examples"
before
EXTRA_QMAKEVARS_PRE += "${@bb.utils.contains('PACKAGECONFIG', 'examples', 'QT_BUILD_PARTS+=examples', '', d)}"
hope it can help you.

cwald, 2020-02-10:
Also encountered compile failure: "No data signature found"
 qt3d 5.12.3+gitAUTOINC+7da5c4c35a

Yocto Project cleansstate and then run default tasks for qt3d, was able to build.  32GB RAM and SWAP space, monitoring, didn't see consumption close to max.
bitbake -c cleansstate qt3d
bitbake qt3d

But not root cause for failure, just a work-a-round.

shr-project, 2020-05-21:
Upstream bug https://bugreports.qt.io/browse/QTBUG-73834