meta-qt5/meta-qt5 #96: qtwebengine compilation fails for armv6

xuhcc, 2018-08-21:
I'm trying to build image using [meta-raspberrypi](https://github.com/agherzan/meta-raspberrypi) and meta-qt5 layers. Target machine is **raspberrypi0-wifi**. Compilation fails with this error (on meta-qt5 master branch):

```
| FAILED: obj/base/base/base_jumbo_37.o
| /home/vagrant/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/qtwebengine/5.11.1+gitAUTOINC+fe73e54057_bad02200c6-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -march=armv6 -mfpu=vfp -mfloat-abi=hard -mtune=arm1176jzf-s -mfpu=vfp --sysroot=/home/vagrant/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/qtwebengine/5.11.1+gitAUTOINC+fe73e54057_bad02200c6-r0/recipe-sysroot -MMD -MF obj/base/base/base_jumbo_37.o.d -DUSE_SYMBOLIZE -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DNO_TCMALLOC -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DCHROMIUM_BUILD -DFIELDTRIAL_TESTING_ENABLED -DTOOLKIT_QT -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DBASE_IMPLEMENTATION -I. -Igen -I../../../../git/src/3rdparty/chromium -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -funwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-missing-field-initializers -Wno-unused-parameter -fno-omit-frame-pointer -g0 -fvisibility=hidden -O2 -fno-ident -fdata-sections -ffunction-sections -std=gnu++14 -fno-delete-null-pointer-checks -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../recipe-sysroot -fvisibility-inlines-hidden -c gen/base/base_jumbo_37.cc -o obj/base/base/base_jumbo_37.o
| {standard input}: Assembler messages:
| {standard input}:2802: Error: selected processor does not support `yield' in ARM mode
```

I tried older versions of Qt (sumo branch), it fails in different place, but the error is the same: `selected processor does not support ``yield' in ARM mode`. Meanwhile, build for raspberrypi3 succeeds.

sanjuchopracool, 2019-01-16:
I also got these errors **only for  raspberrypi0-wifi** when I was building sdk, but I got these errors for qtwebkit.

xuhcc, 2019-01-16:
I did not find a proper solution for this problem and simply removed all pieces of code that caused compilation errors (I believe they are optimizations for ARM).

*qtwebengine_git.bbappend*:

```
SRC_URI_append_armv6 = "\
    file://0001-remove-arm-optimizations.patch;patchdir=src/3rdparty \
    "
```

*0001-remove-arm-optimizations.patch*:

```
diff --git a/chromium/base/allocator/partition_allocator/spin_lock.cc b/chromium/base/allocator/partition_allocator/spin_lock.cc
index f3958fbe30..7a8af94688 100644
--- a/chromium/base/allocator/partition_allocator/spin_lock.cc
+++ b/chromium/base/allocator/partition_allocator/spin_lock.cc
@@ -26,8 +26,6 @@
 #elif defined(COMPILER_GCC) || defined(__clang__)
 #if defined(ARCH_CPU_X86_64) || defined(ARCH_CPU_X86)
 #define YIELD_PROCESSOR __asm__ __volatile__("pause")
-#elif defined(ARCH_CPU_ARMEL) || defined(ARCH_CPU_ARM64)
-#define YIELD_PROCESSOR __asm__ __volatile__("yield")
 #elif defined(ARCH_CPU_MIPSEL)
 // The MIPS32 docs state that the PAUSE instruction is a no-op on older
 // architectures (first added in MIPS32r2). To avoid assembler errors when
diff --git a/chromium/third_party/ffmpeg/libavcodec/arm/Makefile b/chromium/third_party/ffmpeg/libavcodec/arm/Makefile
index 1eeac5449e..442d73f741 100644
--- a/chromium/third_party/ffmpeg/libavcodec/arm/Makefile
+++ b/chromium/third_party/ffmpeg/libavcodec/arm/Makefile
@@ -75,8 +75,7 @@ ARMV6-OBJS-$(CONFIG_MPEGAUDIODSP)      += arm/mpegaudiodsp_fixed_armv6.o
 ARMV6-OBJS-$(CONFIG_MPEGVIDEOENC)      += arm/mpegvideoencdsp_armv6.o
 ARMV6-OBJS-$(CONFIG_PIXBLOCKDSP)       += arm/pixblockdsp_armv6.o
 ARMV6-OBJS-$(CONFIG_STARTCODE)         += arm/startcode_armv6.o
-ARMV6-OBJS-$(CONFIG_VP8DSP)            += arm/vp8_armv6.o               \
-                                          arm/vp8dsp_init_armv6.o       \
+ARMV6-OBJS-$(CONFIG_VP8DSP)            += arm/vp8dsp_init_armv6.o       \
                                           arm/vp8dsp_armv6.o
 
 # decoders/encoders
diff --git a/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h b/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
index feb1247916..7fc8d85b0e 100644
--- a/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
+++ b/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
@@ -37,85 +37,4 @@
 #   define U(x)
 #endif
 
-#if HAVE_ARMV6_INLINE
-
-#define vp56_rac_get_prob vp56_rac_get_prob_armv6
-static inline int vp56_rac_get_prob_armv6(VP56RangeCoder *c, int pr)
-{
-    unsigned shift     = ff_vp56_norm_shift[c->high];
-    unsigned code_word = c->code_word << shift;
-    unsigned high      = c->high << shift;
-    unsigned bit;
-
-    __asm__ ("adds    %3,  %3,  %0           \n"
-             "itt     cs                     \n"
-             "cmpcs   %7,  %4                \n"
-           L("ldrcsh  %2,  [%4], #2          \n")
-           U("ldrhcs  %2,  [%4], #2          \n")
-             "rsb     %0,  %6,  #256         \n"
-             "smlabb  %0,  %5,  %6,  %0      \n"
-           T("itttt   cs                     \n")
-             "rev16cs %2,  %2                \n"
-           T("lslcs   %2,  %2,  %3           \n")
-           T("orrcs   %1,  %1,  %2           \n")
-           A("orrcs   %1,  %1,  %2,  lsl %3  \n")
[... 78 lines cut ...]
-        return y;
-    }
-    inline SkFixed SkFixedMul_arm(SkFixed x, SkFixed y)
-    {
-        int32_t t;
-        asm("smull  %0, %2, %1, %3          \n"
-            "mov    %0, %0, lsr #16         \n"
-            "orr    %0, %0, %2, lsl #16     \n"
-            : "=r"(x), "=&r"(y), "=r"(t)
-            : "r"(x), "1"(y)
-            :
-            );
-        return x;
-    }
-    #undef SkFixedMul
-    #define SkFixedMul(x, y)        SkFixedMul_arm(x, y)
-
-    #undef SkFloatToFixed
-    #define SkFloatToFixed(x)  SkFloatToFixed_arm(x)
-#endif
-
 ///////////////////////////////////////////////////////////////////////////////
 
 #define SkFixedToScalar(x)          SkFixedToFloat(x)
```

nandra, 2019-10-02:
hi any update on this? I'm using now 5.12 from warrior and hit same issue (also patch doesn't apply). Ideas? Thanks.

xuhcc, 2019-10-02:
@nandra 

Here's my version of the patch for `warrior` branch:

```patch
diff --git a/chromium/base/allocator/partition_allocator/spin_lock.cc b/chromium/base/allocator/partition_allocator/spin_lock.cc
index fd062c33b4..0aecee01a1 100644
--- a/chromium/base/allocator/partition_allocator/spin_lock.cc
+++ b/chromium/base/allocator/partition_allocator/spin_lock.cc
@@ -28,8 +28,6 @@
 #elif defined(COMPILER_GCC) || defined(__clang__)
 #if defined(ARCH_CPU_X86_64) || defined(ARCH_CPU_X86)
 #define YIELD_PROCESSOR __asm__ __volatile__("pause")
-#elif (defined(ARCH_CPU_ARMEL) && __ARM_ARCH >= 6) || defined(ARCH_CPU_ARM64)
-#define YIELD_PROCESSOR __asm__ __volatile__("yield")
 #elif defined(ARCH_CPU_MIPSEL)
 // The MIPS32 docs state that the PAUSE instruction is a no-op on older
 // architectures (first added in MIPS32r2). To avoid assembler errors when
diff --git a/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h b/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
index feb1247916..7fc8d85b0e 100644
--- a/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
+++ b/chromium/third_party/ffmpeg/libavcodec/arm/vp56_arith.h
@@ -37,85 +37,4 @@
 #   define U(x)
 #endif
 
-#if HAVE_ARMV6_INLINE
-
-#define vp56_rac_get_prob vp56_rac_get_prob_armv6
-static inline int vp56_rac_get_prob_armv6(VP56RangeCoder *c, int pr)
-{
-    unsigned shift     = ff_vp56_norm_shift[c->high];
-    unsigned code_word = c->code_word << shift;
-    unsigned high      = c->high << shift;
-    unsigned bit;
-
-    __asm__ ("adds    %3,  %3,  %0           \n"
-             "itt     cs                     \n"
-             "cmpcs   %7,  %4                \n"
-           L("ldrcsh  %2,  [%4], #2          \n")
-           U("ldrhcs  %2,  [%4], #2          \n")
-             "rsb     %0,  %6,  #256         \n"
-             "smlabb  %0,  %5,  %6,  %0      \n"
-           T("itttt   cs                     \n")
-             "rev16cs %2,  %2                \n"
-           T("lslcs   %2,  %2,  %3           \n")
-           T("orrcs   %1,  %1,  %2           \n")
-           A("orrcs   %1,  %1,  %2,  lsl %3  \n")
-             "subcs   %3,  %3,  #16          \n"
-             "lsr     %0,  %0,  #8           \n"
-             "cmp     %1,  %0,  lsl #16      \n"
-             "ittte   ge                     \n"
-             "subge   %1,  %1,  %0,  lsl #16 \n"
-             "subge   %0,  %5,  %0           \n"
-             "movge   %2,  #1                \n"
-             "movlt   %2,  #0                \n"
-             : "=&r"(c->high), "=&r"(c->code_word), "=&r"(bit),
-               "+&r"(c->bits), "+&r"(c->buffer)
-             : "r"(high), "r"(pr), "r"(c->end - 1),
-               "0"(shift), "1"(code_word)
-             : "cc");
-
-    return bit;
-}
-
-#define vp56_rac_get_prob_branchy vp56_rac_get_prob_branchy_armv6
-static inline int vp56_rac_get_prob_branchy_armv6(VP56RangeCoder *c, int pr)
-{
-    unsigned shift     = ff_vp56_norm_shift[c->high];
-    unsigned code_word = c->code_word << shift;
-    unsigned high      = c->high << shift;
-    unsigned low;
-    unsigned tmp;
-
-    __asm__ ("adds    %3,  %3,  %0           \n"
-             "itt     cs                     \n"
-             "cmpcs   %7,  %4                \n"
-           L("ldrcsh  %2,  [%4], #2          \n")
-           U("ldrhcs  %2,  [%4], #2          \n")
-             "rsb     %0,  %6,  #256         \n"
-             "smlabb  %0,  %5,  %6,  %0      \n"
-           T("itttt   cs                     \n")
-             "rev16cs %2,  %2                \n"
-           T("lslcs   %2,  %2,  %3           \n")
-           T("orrcs   %1,  %1,  %2           \n")
-           A("orrcs   %1,  %1,  %2,  lsl %3  \n")
-             "subcs   %3,  %3,  #16          \n"
-             "lsr     %0,  %0,  #8           \n"
-             "lsl     %2,  %0,  #16          \n"
-             : "=&r"(low), "+&r"(code_word), "=&r"(tmp),
-               "+&r"(c->bits), "+&r"(c->buffer)
-             : "r"(high), "r"(pr), "r"(c->end - 1), "0"(shift)
-             : "cc");
-
-    if (code_word >= tmp) {
-        c->high      = high - low;
-        c->code_word = code_word - tmp;
-        return 1;
-    }
-
-    c->high      = low;
-    c->code_word = code_word;
-    return 0;
-}
-
-#endif
-
 #endif /* AVCODEC_ARM_VP56_ARITH_H */
```

nandra, 2019-10-02:
@xuhcc thanks a lot

nandra, 2019-10-03:
@xuhcc did you hit also this issue (maybe have fix for that also):
```
y-inlines-hidden -frtti -Wno-narrowing -Wno-attributes -Wno-class-memaccess -Wno-subobject-linkage -c ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/propname.cpp -o v8_snapshot/obj/third_party/icu/bundled_icuuc/propname.o
| In file included from ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/unicode/std_string.h:35:0,
|                  from ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/unicode/stringpiece.h:32,
|                  from ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/unicode/bytestrie.h:26,
|                  from ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/propname.h:18,
|                  from ../../../../git/src/3rdparty/chromium/third_party/icu/source/common/propname.cpp:14:
| /usr/include/c++/6/string:38:28: fatal error: bits/c++config.h: No such file or directory
|  #include <bits/c++config.h>
|                             ^

```
thanks

xuhcc, 2019-10-03:
@nandra No, I didn't.