meta-qt5/meta-qt5 #98: qtsmarthome sources aren't available

gvalcaza, 2018-08-22:
When building the qtsmarthome package, the do_fetch task fails because the tarball containing the sources is no longer available (http://share.basyskom.com/demos/smarthome_src.tar.gz)

gpio-embedded, 2018-08-25:
Is there any different mirror available?

gvalcaza, 2018-08-31:
Not that I know of... The closest thing I could find is this, but the .zip's contents are different: http://d33763z96x0wep.cloudfront.net/wp-content/uploads/2017/04/smarthome.zip

ntwerdochlib, 2021-04-16:
https://artifacts.toradex.com/artifactory/tdxref-oe-prerelease-frankfurt/dunfell-5.x.y/monthly/4/sources/smarthome_src.tar.gz
https://artifacts.toradex.com/artifactory/tdxref-oe-prerelease-frankfurt/dunfell-5.x.y/monthly/4/sources/smarthome_src.tar.gz.md5
https://artifacts.toradex.com/artifactory/tdxref-oe-prerelease-frankfurt/dunfell-5.x.y/monthly/4/sources/smarthome_src.tar.gz.sha1
https://artifacts.toradex.com/artifactory/tdxref-oe-prerelease-frankfurt/dunfell-5.x.y/monthly/4/sources/smarthome_src.tar.gz.sha256

ienecode, 2025-01-03:
https://artifacts.toradex.com/artifactory/tdxref-oe-prod-frankfurt/dunfell-5.x.y/release/4/sources/smarthome_src.tar.gz