meta-qt5/meta-qt5 #108: *.cmake files generated from *.cmake.in files can contain paths to recipe sysroot

martin-ejdestig, 2018-09-10:
Encountered this problem when trying to build https://github.com/KDAB/GammaRay with http://code.qt.io/cgit/yocto/meta-boot2qt.git and https://github.com/agherzan/meta-raspberrypi .

meta-boot2qt sets QMAKE_INCDIR_EGL to extra include directories needed for the VC driver which are in turn added to the Qt5::Gui_GLESv2 target in Qt5GuiConfigExtras.cmake . Gammaray uses the Qt5::Gui_GLESv2 target in https://github.com/KDAB/GammaRay/blob/master/plugins/quickinspector/CMakeLists.txt#L48 . If the qtbase recipe-sysroot does not exist, CMake will fail when configuring Gammaray with the error:

Imported target "Qt5::Gui_GLESv2" includes non-existent path

<path to include directory in qtbase recipe sysroot>

in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
 
   * The path was deleted, renamed, or moved to another location.
 
   * An install or uninstall procedure did not complete successfully.
 
   * The installation package was faulty and references files it does not
   provide.

Something similar to what is done to *.pc and *.prl files, where it is sed:ed out to the empty string, needs to be done for *.cmake files as well.

Unlike for pkg-config and *.prl files though it can not be replaced with the empty string. (The sysroot can be passed to pkg-config through PKG_CONFIG_SYSROOT_DIR and pkg-config will prepend it to all paths when pkg-config extracts them from *.pc. In CMake there is https://cmake.org/cmake/help/latest/variable/CMAKE_SYSROOT.html but it looks like it is not set in Yocto.)

martin-ejdestig, 2018-09-10:
See https://github.com/Pelagicore/meta-pelux/pull/185 for workaround.

kanavin, 2018-09-10:
This is the original upstream file:
https://github.com/qt/qtbase/blob/5.11/src/gui/Qt5GuiConfigExtras.cmake.in
To me, it seems like CMAKE_INCLUDE_DIR_IS_ABSOLUTE needs to be true?

martin-ejdestig, 2018-09-10:
Answered in https://github.com/Pelagicore/meta-pelux/pull/185 but repeating here for convenience:

CMAKE_INCLUDE_DIR_IS_ABSOLUTE is only used if CMAKE_ANGLE_EGL_DLL_RELEASE is set (Windows only, I assume).

No indentation, so hard to follow. But relevant parts for us is after !!ELSE at line 53, if I am not mistaken.

And CMAKE_INCLUDE_DIR_IS_ABSOLUTE is something cooked up by Qt? Does not seem like a standard CMake variable. (Bad of them to invade that namespace if that's the case. At least I cannot find it in the CMake source tree, nor in the online documentation.) So I do not see how it would help. But I may be missing something.