meta-qt5/meta-qt5 #128: SDK

Micha1808, 2018-11-23:
How can I add qmlcachegen-native to SDK (qtdeclarative-native)?

BlackZork, 2019-02-01:
I created sdk using thud branch and qtdeclarative binaries are not installed by SDK install script.
I had to copy them manually after install:

cp ```${BUILDROOT}/tmp/sysroots-components/x86_64/qtdeclarative-native/usr/bin {$SDKROOT}/sysroots/x86_64-pokysdk-linux/usr/bin```

After that my simple qml application was deployed and run from qtcreator on host.
As newbie I had no idea how create a proper fix for it.

jpnurmi, 2019-02-13:
Did anyone find a nice workaround?

BlackZork, 2019-02-13:
It works on master branch with Qt 5.12

jpnurmi, 2019-02-13:
Thanks, what's the correct package/dependency/config/anything to get qmlcachegen on SDK? I'm using the master branch with Qt 5.12, but my SDK is still missing qmlcachegen.

BlackZork, 2019-02-13:
I am using meta-rpi created by Jumpnowtek. After updating meta-qt5 to master problem disapperared.  That's all I know.

desert, 2019-02-14:
I suggest to change the title to something more meaningful like: "SDK does not contain qmlcachegen" and state it is, in fact, needed for anything above 5.9. I stumbled onto this topic and it's an important bug.

desert, 2019-02-14:
My status is the same as @BlackZork btw

desert, 2019-02-14:
@BlackZork solution will only work if the full paths of yocto are on the developers' machine. There are some sed replacements done by the build system which do not occur if you copy/paste it directly.

BlackZork, 2019-02-15:
> It works on master branch with Qt 5.12

Errata: I was wrong - it does not work. It seems that I installed new sdk without removing old files. qtdeclarative nativesdk binaries are still missing in SDK on master branch.

BlackZork, 2019-02-15:
Here is the patch:
```
diff --git a/recipes-qt/packagegroups/nativesdk-packagegroup-qt5-toolchain-host.bb b/recipes-qt/packagegroups/nativesdk-packagegroup-qt5-toolchain-host.bb
index e92cb9d..1a87aef 100644
--- a/recipes-qt/packagegroups/nativesdk-packagegroup-qt5-toolchain-host.bb
+++ b/recipes-qt/packagegroups/nativesdk-packagegroup-qt5-toolchain-host.bb
@@ -11,5 +11,6 @@ RDEPENDS_${PN} += " \
     nativesdk-packagegroup-sdk-host \
     nativesdk-qttools-tools \
     nativesdk-qtbase \
+    nativesdk-qtdeclarative-tools \
     nativesdk-perl-modules \
 "
```

desert, 2019-02-16:
I was going to test your patch, however for thelast 5 hours I've been unable to get rid of qtwebkit. Do you happen to know how? just tested
```
TOOLCHAIN_TARGET_TASK_remove="qtwebkit qtwebkit-dev"
TOOLCHAIN_HOST_TASK_remove  ="qtwebkit qtwebkit-dev"
PACKAGECONFIG_remove_pn-qttools = "qtwebkit qtwebkit-dev"
PACKAGECONFIG_remove_pn-qtquick1 = "qtwebkit qtwebkit-dev"
PACKAGECONFIG_remove_pn-qtbase = "qtwebkit qtwebkit-dev"
```

but keeps pulling it to build (and crashing on `thud` and `master`). If you could come with an idea I would immediatly test your patch.

BlackZork, 2019-02-16:
See https://github.com/BlackZork/meta-rpi/commit/7fb0742a1199b36697ae23668af1a1618fa6b001

desert, 2019-02-16:
Excellent, just patched and relaunched my CI bot (with the patch), as soon as it finishes will inform.

desert, 2019-02-18:
Strangely, I was using 201fcf27cf07d60b7d6ab89c7dcefe2190217745 for Qt 5.11, had no qmlcachegen and then went up to fb71293f257c6fd02ccc06765a96dbf11d4569a0 to test your patch and go up, now I got my eglfs_kms not working.

BlackZork, 2019-02-18:
In your case I would not upgrade QT. Just add this patch or create bbappend file just like here: https://github.com/BlackZork/meta-rpi/commit/9f26e5c757a69ba3310fe6ae00a058df7aebc19a

desert, 2019-02-19:
Well, I can say now i got qmlcachegen, and i can cross compile qt console apps, widget apps and even the simple opengl examples (the one with the old green Qt logo).  

But anything I try to cross compile with the SDK that has qml just crashes at startup. I've a hunch it is related to some bad precompilation of qml files. Do you know a waiy bypass it completely, so I can confirm?

BTW: building the qml apps with bitbake makes them run cleanly, but not with the SDK, Im using 
 201fcf27cf07d60b7d6ab89c7dcefe2190217745 for Qt 5.11.3

desert, 2019-02-19:
- Good news is that I get qmlcachegen included, 
- Bad news is the SDK produces binaries that crash. 
- Good news is that adding 
`qputenv("QML_DISABLE_DISK_CACHE", "true");`
 or simply using that ENV variable makes the cache to be disabled (no more crashes !)
 - Bad news is that this seems to be a bug in Qt 5.11 (maybe others) because it tries to regenerate cache into read only rcc files
 - Good news is that we can avoid it
 - Bad news is that adding that ENV variable disables precompilation of everything, not just rcc bundled files.

finally, good news is that I can ignore it and just return to my normal work because I really don't need precompilation. But I leave it here just for others to save a few hours of their life debugging this.

BlackZork, 2019-02-20:
I do not see any problems with QT qml app cross compiled for RaspberryPI. Maybe you steeped into a bug that is related to your hardware type.

desert, 2019-02-23:
Well, it's not related to the hardware, it is related to a Qt bug before 5.12, I just eliminated the qmcache on runtime and it worked. 

BTW your patch also works good.

thesillywhat, 2019-08-05:
When I try to build my application i get an error. No rule to make target
/opt/poky/2.7.1/sysroots/x86_64-pokysdk-linux/usr/bin/qmlcachegen. 

Is this thread talking about the same issue?

And that patch is the only workaround for this issue?

BlackZork, 2019-08-05:
Yes, this is the same issue.
Patch is not workaround, it is a fix. See [this](https://github.com/meta-qt5/meta-qt5/issues/128#issuecomment-464106207) or [this](https://github.com/BlackZork/meta-rpi/commit/9f26e5c757a69ba3310fe6ae00a058df7aebc19a)