meta-qt5/meta-qt5 #129: qt5-meta-toolchain does not build of no opengl

scottellis, 2018-11-25:
This is the error for a board where opengl is not in DISTRO_FEATURES

```
...
| CMake Error at Source/cmake/OptionsQt.cmake:529 (message):
|   Qt with OpenGL support is required for ENABLE_OPENGL
| Call Stack (most recent call first):
|   Source/cmake/WebKitCommon.cmake:50 (include)
|   CMakeLists.txt:137 (include)

```
The SDK builds okay for boards where opengl is in DISTRO_FEATURES.

Using this patch to get around it.

```
index d020cc3..2f8d8a2 100755
--- a/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb
+++ b/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb
@@ -9,7 +9,7 @@ PACKAGEGROUP_DISABLE_COMPLEMENTARY = "1"

 # Requires Ruby to work
 USE_RUBY = " \
-    qtwebkit-dev \
+    ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qtwebkit-dev', '', d)} \
 "

 # Requires Wayland to work

```

otavio, 2018-11-25:
It makes sense, it would be good if you could send a PR doing this change. 

Looking forward to review it :+1:

closed 2018-11-26 by otavio

otavio, 2018-11-26:
You can send a PR to backport it. Please do.

PR #130 (2018-11-26): packagegroup-qt5-toolchain-target: Skip qtwebkit if opengl not found
Resolves #129
commit fe453ec4 (2018-11-26): packagegroup-qt5-toolchain-target: Skip qtwebkit if opengl not found

Resolves #129