meta-qt5/meta-qt5 #134: error on bitbake qtwebengine

likeaferrari, 2018-12-05:
Hello.
When bitbake qtwebengine, it occurs error.
target system is arm7i architecture, and my host system is ubuntu 16.04
here's the log msg.

> ERROR at //build/config/android/internal_rules.gni:12:1: Assertion failed.
> assert(is_android)
> ^-----
> See //build/config/android/rules.gni:8:1: whence it was imported.
> import("//build/config/android/internal_rules.gni")
> ^-------------------------------------------------
> See //third_party/android_tools/BUILD.gn:5:1: whence it was imported.
> import("//build/config/android/rules.gni")
> ^----------------------------------------
> See //third_party/openmax_dl/dl/BUILD.gn:199:16: which caused the file to be included.
>       deps = [ "//third_party/android_tools:cpu_features" ]
>                ^-----------------------------------------
> Project ERROR: GN run error!
> ERROR: oe_runmake failed
> Makefile:80: recipe for target 'sub-gn_run-pro-make_first' failed
> make[2]: *** [sub-gn_run-pro-make_first] Error 3
> make[2]: Leaving directory '/home/ku/yocto/build/tmp/work/armv7ahf-vfp-poky-linux-gnueabi/qtwebengine/5.10.1+gitAUTOINC+9dc8dff7a8_c858cc7609-r0/build/src/core'
> Makefile:78: recipe for target 'sub-core-make_first' failed
> make[1]: *** [sub-core-make_first] Error 2
> make[1]: Leaving directory '/home/ku/yocto/build/tmp/work/armv7ahf-vfp-poky-linux-gnueabi/qtwebengine/5.10.1+gitAUTOINC+9dc8dff7a8_c858cc7609-r0/build/src'
> Makefile:46: recipe for target 'sub-src-make_first' failed
> make: *** [sub-src-make_first] Error 2
> WARNING: exit code 1 from a shell command.
> ERROR: Function failed: do_compile (log file is located at /home/ku/yocto/build/tmp/work/armv7ahf-vfp-poky-linux-gnueabi/qtwebengine/5.10.1+gitAUTOINC+9dc8dff7a8_c858cc7609-r0/temp/log.do_compile.29696)

otavio, 2018-12-13:
Hello @likeaferrari, this error is strange. It is likely related to the BSP you are using. 

What is the BSP and branch that you are using?

likeaferrari, 2018-12-14:
Hello otavio.

I'm on yocto Rocko, and meta-qt5's branch is  master commit e41caf4
Because of the past works, I'm in the situation that have to use Qt 5.10.x base.

I soved this problem in expedient way,

1. Followed QT Official Guide

 https://wiki.qt.io/QtWebEngine/How_to_Try

I tried the guide the link above.

2. Patch "qtwebengine/src/3rdparty/chromium/third_party/openmax_dl/dl/BUILD.gn"

I found the patch work like below, 

https://github.com/voidlinux/void-packages/pull/8002/commits/b43e7ccc137050253b8ec777712adbe778583791#diff-cfa1d6e7ace67def8960a7d644e789b1R31

 and when I apply "srcpkgs/qt5/patches/arm-void-is-not-android.patch" in the link, then the errors gone.

PLUS, If you use Virtual Machine, It may have the memory more than 4GB, because "virtual memory exhausted: Cannot allocate memory" when compile QtWebEngine.

otavio, 2018-12-14:
Hello @likeaferrari 

> I'm on yocto Rocko, and meta-qt5's branch is master commit [e41caf4](https://github.com/meta-qt5/meta-qt5/commit/e41caf42e9e8a69d2f8f7a70365fbb441f0a01e8)
> Because of the past works, I'm in the situation that have to use Qt 5.10.x base.

I see; it would be good if you could try the current master branch to check if this issue is still happening and if so, we could work on a fix for it.

>     1. Patch "qtwebengine/src/3rdparty/chromium/third_party/openmax_dl/dl/BUILD.gn"
> 
> I found the patch work like below,
> 
> [voidlinux/void-packages@b43e7cc#diff-cfa1d6e7ace67def8960a7d644e789b1R31](https://github.com/voidlinux/void-packages/commit/b43e7ccc137050253b8ec777712adbe778583791#diff-cfa1d6e7ace67def8960a7d644e789b1R31)
> 
> and when I apply "srcpkgs/qt5/patches/arm-void-is-not-android.patch" in the link, then the errors gone.

This is indeed a good news; would you mind giving a go in current master branch so we can try to fix it for current releases?

closed 2018-12-26 by likeaferrari