meta-qt5/meta-qt5 #136: Qt 5.12 update

bachp, 2018-12-10:
[Qt 5.12 is the new LTS](http://blog.qt.io/blog/2018/12/06/qt-5-12-lts-released/) release. The goal of this issue is to track the progress of upgrading meta-qt5 to Qt 5.12.

desert, 2018-12-17:
Is it possible it could just need to switch commit SHA hashes and give it a go? If so, I can test it if someone can point me to those sha's.

desert, 2018-12-19:
Which new module? If so, maybe we can build everything and give it a go, Qt modules are independent by nature, unless there's something im missing.

migronof, 2018-12-20:
latest 5.12 is quite broken presently. There are exclusive commits now in both 5.12 & 5.12.1 branches in varying amounts depending on module. I'll try to compose an 'upgrade to 5.12.0' commit to master now that I got 5.11.3 done,

shr-project, 2018-12-20:
https://github.com/meta-qt5/meta-qt5/pull/143

commit b7028b4b (2018-12-20): qt5: upgrade to latest revision in 5.12

* WIP
* fixes #136
commit c6be23d3 (2018-12-21): qt5: upgrade to latest revision in 5.12

* WIP: only qtbase now
* fixes #136
closed 2018-12-26 by shr-project