meta-qt5/meta-qt5 #147: Error on linux firmware 4.9

peel70, 2019-01-04:


peel70, 2019-01-04:
Hello,
I'm building Bitbake images for intel-corei7-64 for old Intel Processors and I've to use linux intel firmware 4.9 
I've noticed that the brand THUD a few days ago was shipped with QT 5.11.3 (before was QT5.11.2)
After this branch update I'm not able to launch qt apps on the target computer.

Launching my test app I've got this error:

./Test01: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory

after some investigations using the command ldconfig -p I've noticed this:

libQt5Core.so.5 (libc6,x86-64, OS ABI: Linux 4.11.0) => /usr/lib64/libQt5Core.so.5
libQt5Core.so (libc6,x86-64, OS ABI: Linux 4.11.0) => /usr/lib64/libQt5Core.so
 
Is the only QT5 library that have OS ABI configured.
It mean that the minimum linux firmare is 4.11?

Searching on internet I've found this

strip --remove-section=.note.ABI-tag /usr/lib64/libQt5Core.so

I have not try at the moment, but I don't understand why an update of the Qt on the same branch.

Is there a fix?
Regards
Enrico

peel70, 2019-01-04:
Hello,
I've tested master branch with the Qt5.12.0 and the problem is the same.

So I've tried with meta-qt5 THUD's branch with the command 

git fetch origin 010a29e0ab5eb2a7efde98ee0a4f78884281bc09

this build Qt5 5.11.2

now ldconfig -p print:

libQt5Core.so.5 (libc6,x86-64, OS ABI: Linux 3.16.0) => /usr/lib64/libQt5Core.so.5
libQt5Core.so (libc6,x86-64, OS ABI: Linux 3.16.0) => /usr/lib64/libQt5Core.so

and my application is working again.

Please let me know if this issue will be fixed or if I've to use 5.11.2Qt for intel linux 4.9 firmware

regards
Enrico

shr-project, 2019-01-04:
Hi,

try to find out which qtbase commit caused this and report it to Qt upstream, I'm not using Intel for anything, so I'm not familiar with their firmware limitations.

peel70, 2019-01-07:
Hello,
I'll try in the next days and keep you informed.

peel70, 2019-04-02:
Hello,
it seem to be fixed in the latest Yocto's Thud branch but linux firmware 4.9 is not supported but the computer now work with 4.19.

Regards
Enrico

closed 2019-04-02 by peel70