meta-qt5/meta-qt5 #159: qtwebengine requires 32 bit host environment

ensc, 2019-01-28:
When trying to build qtwebengine on a 64 bit system for ARM, bitbake fails with

```
| ...hosttools/g++ ...  -m32 -msse2 -mfpmath=sse -mmmx -o v8_snapshot/obj/third_party/icu/bundled_icui18n/decContext.o
| In file included from /usr/include/features.h:452,
|                  from /usr/include/bits/libc-header-start.h:33,
|                  from /usr/include/string.h:26,
|                  from ../../../../git/src/3rdparty/chromium/third_party/icu/source/i18n/decContext.cpp:25:
| /usr/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-32.h: No such file or directory
|  # include <gnu/stubs-32.h>
```

`src/buildtools/configure_host.pro` causes the 64 to 32 bit "downgrade" by

```
contains(GN_TARGET_CPU, "arm")|contains(GN_TARGET_CPU, "mipsel")|contains(GN_TARGET_CPU, "x86") {
    # The v8 snapshot need a host that matches bitwidth, so we build makesnapshot to 32-bit variants of host.
    contains(GN_V8_HOST_CPU, x64): GN_V8_HOST_CPU = "x86"
```

Seen in "thud" branch; rev 201fcf27cf07d60b7d6ab89c7dcefe2190217745

nros, 2019-03-28:
On CentOS 7, you need to install the package `glibc-devel.i686`.
```
 yum install glibc-devel.i686
```
That will install the missing header file. Other distros have similar packages, like `libc-dev:i386` in Debian.

schnitzeltony, 2019-03-28:
FWIW: on master you need glibc-static either (think 32Bit - installed both 32/64 so I don't know). Maybe it is worth dropping a note in the recipe?

schnitzeltony, 2019-03-29:
Just had it a my home PC and I was wrong: it is not glib but libsstdc++-static which is required

| /home/superandy/tmp/oe-core-glibc/hosttools/ld: cannot find -lstdc++

ensc, 2019-03-29:
I think this is a really bad situation and violates the self-hosting idea behind OpenEmbedded.  Perhaps this code can be compiled for the target and executed with qemu?

shr-project, 2019-03-29:
It's like this since 5.9 upgrade, see:
https://github.com/meta-qt5/meta-qt5/commit/969f1f80bf255498abbec6886d443670c20a79c8
I don't like it at all, see:
http://lists.openembedded.org/pipermail/openembedded-devel/2017-August/114417.html
and similar discussions from that time, but in the end I gave up, if Qt isn't going to ever fix it, I don't want to carry another not-upstream-able patch in meta-qt5 (there are too many already).

Please send README update.