meta-qt5/meta-qt5 #176: qtbase-native do_configure fails  without zlib1g-dev on the build host (5.12 branch)

lucaceresoli, 2019-03-11:
qtbase-native_git.bb:do_configure fails when `zlib1g-dev` is not installed on the PC.

My setup:
 * Ubuntu 18.04 build host
 * `zlib1g-dev` not installed
 * poky thud
 * meta-qt5 5.12

The do_configure step fails with:

> Command line: -opensource -confirm-license -sysroot /home/ceresoli/temp/bug-meta-qt5/poky/build/tmp/work/x86_64-linux/qtbase-native/5.12.1+gitAUTOINC+0c49815333-r0/recipe-sysroot-native -no-gcc-sysroot -system-zlib [...]
> ...
> Checking for zlib... 
> Trying source 0 (type inline) of library zlib ...
>   => source failed condition 'config.msvc'.
> Trying source 1 (type inline) of library zlib ...
>   => source failed condition 'config.msvc'.
> Trying source 2 (type inline) of library zlib ...
> None of [libz.so libz.a] found in [] and global paths.
>   => source produced no result.
> Trying source 3 (type inline) of library zlib ...
>   => source failed condition 'config.wasm'.
> test config.qtbase.libraries.zlib FAILED
> ...
>   Using system zlib ...................... no
> ...
> ERROR: Feature 'system-zlib' was enabled, but the pre-condition 'libs.zlib' failed.

lucaceresoli, 2019-03-11:
Script to reproduce the issue. Run after `sudo apt remove zlib1g-dev`.

```bash
#!/bin/bash

set -e
set -u
set -x

rm -fr poky

POKY_REF=""
if [ -d ref-poky.git ]; then
    POKY_REF="--reference ref-poky.git"
fi    
git clone -b thud ${POKY_REF} git://git.yoctoproject.org/poky
cd poky
git checkout b904775c2b82f110a9e0ae9281be452546916fea # fails with our without this line

QT5_REF=""
if [ -d ../ref-meta-qt5.git ]; then
    QT5_REF="--reference ../ref-meta-qt5.git"
fi    
git clone -b 5.12 ${QT5_REF} git://code.qt.io/yocto/meta-qt5.git
git -C meta-qt5 checkout v5.12.1 # fails with our without this line

set +u
source oe-init-build-env
set -u

if [ -d /opt/bitbake-downloads ]; then
    echo "DL_DIR=\"/opt/bitbake-downloads\"" >>conf/local.conf
fi    

bitbake-layers add-layer  ../meta-qt5/

bitbake qtbase-native

```

shr-project, 2019-04-11:
see https://github.com/meta-qt5/meta-qt5/pull/181

closed 2019-04-11 by shr-project