meta-qt5/meta-qt5 #183: Qt5WebEngineCore.pc: No such file or directory

xuhcc, 2019-04-03:
Setup:

* meta-qt5 + [meta-odroid](https://github.com/akuster/meta-odroid/tree/thud)
* **thud** branch
* `DISTRO_FEATURES_append = " x11 opengl "`

When I'm trying to build qtwebengine, the following error occurs:

```
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/vagrant/meta-qt5/recipes-qt/qt5/qtdeclarative_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-support/libcap/libcap_2.25.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-graphics/xorg-lib/pixman_0.34.0.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-devtools/gcc/gcc-runtime_8.2.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-support/icu/icu_62.1.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-support/libevent/libevent_2.1.8.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtxmlpatterns_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-multimedia/libpng/libpng_1.6.35.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-graphics/drm/libdrm_2.4.94.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-devtools/bison/bison_3.0.4.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-devtools/ninja/ninja_1.8.2.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-support/nss/nss_3.39.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-graphics/pango/pango_1.42.4.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-devtools/nasm/nasm_2.13.03.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-extended/gperf/gperf_3.1.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-core/glibc/glibc_2.28.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-bsp/pciutils/pciutils_3.6.2.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtquickcontrols_git.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-multimedia/libpng/libpng_1.6.35.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtbase-native_git.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-multimedia/alsa/alsa-lib_1.1.6.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-graphics/fontconfig/fontconfig_2.12.6.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtlocation_git.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-core/gettext/gettext_0.19.8.1.bb:do_populate_sysroot', 'virtual:native:/home/vagrant/poky/meta/recipes-support/nspr/nspr_4.19.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-connectivity/openssl/openssl_1.1.1a.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtquickcontrols2_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-devtools/python/python-native_2.7.15.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-devtools/gcc/gcc-cross_8.2.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-graphics/cairo/cairo_1.14.12.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-support/nss/nss_3.39.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtwebchannel_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-devtools/perl/perl-native_5.24.4.bb:do_populate_sysroot', '/home/vagrant/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_populate_sysroot', '/home/vagrant/poky/meta/recipes-multimedia/webp/libwebp_1.0.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['qtdeclarative', 'libcap', 'pixman', 'gcc-runtime', 'icu', 'libevent', 'qtxmlpatterns', 'libpng', 'libdrm', 'bison-native', 'ninja-native', 'nss-native', 'pango', 'nasm-native', 'gperf-native', 'glibc', 'pciutils', 'qtquickcontrols', 'libpng-native', 'qtbase-native', 'pseudo-native', 'alsa-lib', 'fontconfig', 'qtlocation', 'gettext-native', 'nspr-native', 'openssl', 'qtquickcontrols2', 'python-native', 'gcc-cross-aarch64', 'quilt-native', 'cairo', 'nss', 'qtwebchannel', 'perl-native', 'qtbase', 'libwebp', 'zlib', 'util-macros', 'libgcc', 'linux-libc-headers', 'opkg-utils', 'libpciaccess', 'libpthread-stubs', 'gnu-config-native', 'libtool-native', 'gettext-minimal-native', 'autoconf-native', 'xz-native', 'automake-native', 'texinfo-dummy-native', 'flex-native', 're2c-native', 'sqlite3-native', 'zlib-native', 'gobject-introspection', 'harfbuzz', 'fribidi', 'libxft', 'libx11', 'glib-2.0', 'freetype', 'groff-native', 'kmod', 'pkgconfig-native', 'dbus-native', 'expat', 'expat-native', 'openssl-native', 'readline-native', 'db-native', 'bzip2-native', 'gdbm-native', 'mpfr-native', 'gmp-native', 'libmpc-native', 'binutils-cross-aarch64', 'libxrender', 'mesa-gl', 'libxext', 'libxcb', 'libsm', 'sqlite3', 'nspr', 'qtwebsockets', 'dbus', 'libjpeg-turbo', 'libxkbcommon', 'libxi', 'xcb-util-renderutil', 'eudev', 'xcb-util-keysyms', 'xcb-util-image', 'xcb-util-wm', 'm4-native', 'python3', 'libffi', 'autoconf-archive', 'xorgproto', 'xtrans', 'util-linux', 'libpcre', 'bash-completion', 'ncurses-native', 'libxfixes', 'libxdamage', 'libxxf86vm', 'libxau', 'xcb-proto', 'libxdmcp', 'libice', 'update-rc.d', 'shadow-native', 'shadow-sysroot', 'base-passwd', 'base-files', 'shadow', 'initscripts', 'xkeyboard-config', 'xcb-util', 'xz', 'bzip2', 'gdbm', 'libxcrypt', 'readline', 'ncurses', 'attr']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| NOTE: make -j 4 MAKEFLAGS=-j 4 OE_QMAKE_COMPILER=aarch64-poky-linux-gcc  -fstack-protector-strong  -D_FORTIFY_SOURCE=2  --sysroot=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot OE_QMAKE_CC=aarch64-poky-linux-gcc  -fstack-protector-strong  -D_FORTIFY_SOURCE=2  --sysroot=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot OE_QMAKE_CXX=aarch64-poky-linux-g++  -fstack-protector-strong  -D_FORTIFY_SOURCE=2  --sysroot=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0=/usr/src/debug/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0 -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot= -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot-native=  OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0=/usr/src/debug/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0 -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot= -fdebug-prefix-map=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden OE_QMAKE_LINK=aarch64-poky-linux-g++  -fstack-protector-strong  -D_FORTIFY_SOURCE=2  --sysroot=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now OE_QMAKE_AR=aarch64-poky-linux-ar OE_QMAKE_STRIP=echo OE_QMAKE_INCDIR_QT=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/recipe-sysroot//usr/include install INSTALL_ROOT=/home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/image
| make: Nothing to be done for 'install'.
| sed: can't read /home/vagrant/build/tmp/work/aarch64-poky-linux/qtwebengine/5.11.3+gitAUTOINC+18412af977_2095a35f30-r0/image/usr/lib/pkgconfig/Qt5WebEngineCore.pc: No such file or directory
```

For some reason compilation is being skipped by bitbake, and then `do_install` fails. I found this question on SO: https://stackoverflow.com/questions/36143564/qt-webengine-5-6-recipe-do-install but the answer didn't help.

shr-project, 2019-04-03:
Read the do_configure and do_compile log, my guess is that you're missing one of mandatory PACKAGECONFIGs for qtbase as mentioned in README.

xuhcc, 2019-04-03:
Thanks for advice! Indeed, there is a message in do_configure log:
```
khronos development headers appear to be missing (mesa/libegl1-mesa-dev)
QtWebEngine will not be built.
```
But I can't figure out what PACKAGECONFIGs are needed.

CleberPeter, 2019-05-06:
Same issue here, does anyone have any suggestions?

xuhcc, 2019-05-06:
@CleberPeter 

You could try to add `PACKAGECONFIG_GL = "gles2"` to `qtbase_git.bbappend`.

With this config **qtwebengine** builds successfully in my envrionment, however I ran into other issues so can't actually confirm whether it works or not.