meta-qt5/meta-qt5 #197: sdk generationg Postinstall failing on qtbase-tools due to rescent commit

rarengifoArthrex, 2019-04-29:
Bad commit below has caused sdk generation to fail on the post install with the error stated below. 
Working - bd4b0cd2ccf02bb544602739da8aae1c2a6638a2
Bad - 7fdddb66587074c8f3f7b839054303a73c707e36

ERROR: arthrex-v2-base-qt-image-1.0-r0 do_populate_sdk: Postinstall scriptlets of ['qtbase-tools'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/rrengifo/yocto/nano/poky/build-nanoscope/tmp/work/nanoscope-oe-linux-gnueabi/arthrex-v2-base-qt-image/1.0-r0/temp/log.do_populate_sdk.
ERROR: arthrex-v2-base-qt-image-1.0-r0 do_populate_sdk: Function failed: do_populate_sdk
ERROR: Logfile of failure stored in: /home/rrengifo/yocto/nano/poky/build-nanoscope/tmp/work/nanoscope-oe-linux-gnueabi/arthrex-v2-base-qt-image/1.0-r0/temp/log.do_populate_sdk.45247
ERROR: Task (/home/rrengifo/yocto/nano/poky/build-nanoscope/../meta-act/arthrex/images/arthrex-v2-base-qt-image.bb:do_populate_sdk) failed with exit code '1'

kekiefer, 2019-05-02:
I've found the same thing. It looks like pkg_postinst_${PN}-tools is expecting /usr/bin/qt.conf, when the file is actually at /usr/bin/qt5/qt.conf.

It seems to me like it should be looking for the file in the same manner it was created, by interpolating `OE_QMAKE_PATH_BINS`.

The following fixed the issue for me. I can submit a PR if someone more knowledgable in such matters can confirm this is the correct fix.

```
From e5c3b23e8830ba982e20278f6581d59b21b0874d Mon Sep 17 00:00:00 2001
From: Kurt Kiefer <kekiefer@gmail.com>
Date: Thu, 2 May 2019 11:39:39 -0700
Subject: [PATCH] qtbase-tools: Fix error during pkg_postinst

When staging the sdk, the following error is seen:
sed: can't read <path-to-work>/sdk/image/usr/local/oecore-x86_64/sysroots/aarch64-oe-linux/usr/bin/qt.conf: No such file or directory
because the qt.conf file moved from /usr/bin/qt.conf to
/usr/bin/qt5/qt.conf

This change locates the qt.conf file during pkg_postinst using the
OE_QMAKE_PATH_BINS variable, just as when the file is created.
---
 recipes-qt/qt5/qtbase_git.bb | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/recipes-qt/qt5/qtbase_git.bb b/recipes-qt/qt5/qtbase_git.bb
index 99adbb6b..8fb8b405 100644
--- a/recipes-qt/qt5/qtbase_git.bb
+++ b/recipes-qt/qt5/qtbase_git.bb
@@ -282,7 +282,7 @@ pkg_postinst_${PN}-tools () {
 sed -i \
     -e 's:HostSpec =.*:HostSpec = ${TARGET_MKSPEC}:g' \
     -e 's:TargetSpec =.*:TargetSpec = ${TARGET_MKSPEC}:g' \
-    $D${bindir}/qt.conf
+    $D${OE_QMAKE_PATH_BINS}/qt.conf
 }
 
 pkg_postinst_${PN}-mkspecs () {
-- 
2.17.1
```

shr-project, 2019-05-02:
The fix looks reasonable, please send PR and lets ask @schnitzeltony if it still works for him.