meta-qt5/meta-qt5 #198: switch -mcpu=cortex-a9 conflicts with -march=armv7-a switch

h4nn35, 2019-04-30:
Tried my fist warrior build and saw a lot of warnings like this:
"cc1plus: warning: switch -mcpu=cortex-a9 conflicts with -march=armv7-a switch" in the build log.
I suspect that this is related to oe-core change "arm-tunes: Remove -march option if mcpu is already added".
There seems to be a fix for this in meta-browser: "chromium-gn: Computer arm_version from overrides" by @kraj

h4nn35, 2019-05-03:
I've added the mentioned patch to my build and as far as I could check, there are no more warnings about conflicts between march and mcpu in the logs.
Would this patch be appropriate for qtwebengine?

kraj, 2019-05-03:
Yes please submit a pull request

closed 2019-05-13 by h4nn35