meta-qt5/meta-qt5 #218: module "QtMultimedia" is not installed

QB4-dev, 2019-07-13:
Embedded linux board with iMX6ULL, I added following items to my image:
```
   qtbase \
   qtbase-plugins \
   qtserialport \
   qt5-env \
   libxkbcommon \
   qtquickcontrols \
   qtquickcontrols2 \
   qtmultimedia \
   qtmultimedia-plugins \
   qtmultimedia-qmlplugins \
   gstreamer1.0 \
   gstreamer1.0-libav \
   gstreamer1.0-plugins-good \
   gstreamer1.0-plugins-bad \
   gstreamer1.0-plugins-imx \
```
Building **qtbase** with:
```
PACKAGECONFIG_append = " no-opengl accessibility fontconfig linuxfb libinput xkbcommon"
PACKAGECONFIG_remove = " examples tests gbm eglfs gl gles2 kms"
```
and **qtmultimedia** with:
```
PACKAGECONFIG_append = "gstreamer"
```
camera exmaple with Qt5 widgets works properly.
declarative camera demo or any other QML example with **multimedia** module doesn't works because of following error:
```
QQmlApplicationEngine failed to load component
qrc:/main.qml:3 module "QtMultimedia" is not installed
```
In **/usr/lib/qml** there are no QtMultimedia related files:
```
> ls /usr/lib/qml/
Qt  QtQml  QtQuick  QtQuick.2  QtTest  builtins.qmltypes
```
also
```
>  opkg files qtmultimedia-qmlplugins
Package qtmultimedia-qmlplugins (5.12.3+git0+64981aee3b-r0) is installed on root and has the following files:
```
returns no files installed.
How to build QML multimedia plugins with Yocto?

G-S-E, 2020-03-02:
Do you have any luck with the issue?

QB4-dev, 2020-03-02:
@G-S-E unfortunately no. According to my investigations qtmultimedia qml plugins needs opengl support to be compiled. 
In the end I found way to connect to QML web camera image preview that I needed using gstreamer videosink from this project:
https://github.com/muddog/QRScanner
and rendering it as QML element using QPainter like in this example code
https://doc.qt.io/qt-5/qtqml-tutorials-extending-qml-example.html

sateesh0457, 2020-11-13:
add in your conf/local.conf with latest version 5.15
IMAGE_INSTALL_append = " qtmultimedia"
depends on DISTRO_FEATURES_append = " opengl"