meta-qt5/meta-qt5 #245: Cross compiling qttools for RISC-V uses host glibc

alistair23, 2019-10-16:
When building qttools for RISC-V 64-bit I see all sorts of float128 errors like this:

> | /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/qttools/5.13.0+gitAUTOINC+cc9250477e-r0/recipe-sysroot/usr/include/stdlib.h:152:8: error: '_Float128' does not name a type; did you mean '_Float32x'?
> |   152 | extern _Float128 strtof128 (const char *__restrict __nptr,
> |       |        ^~~~~~~~~
> |       |        _Float32x
> | /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/qttools/5.13.0+gitAUTOINC+cc9250477e-r0/recipe-sysroot/usr/include/stdlib.h:245:4: error: '_Float128' has not been declared
> |   245 |    _Float128 __f)
> |       |    ^~~~~~~~~

These errors come from the glibc headers and not qttools itself. Looking at the glibc headers all of these float128 defines are protected by

```#if __HAVE_FLOAT128```

Which is not set for RISC-V, but is set for x86.

It seems like the host libc libraries are being used when building qttools.

RealLitb, 2019-10-22:
The Qt layer uses the "linux-oe-g++" qmake platform specification (see qmake5_base bbclass), which includes "qtbase/mkspecs/linux-g++/qplatformdefs.h" (see https://github.com/meta-qt5/meta-qt5/blob/zeus/recipes-qt/qt5/qtbase/0001-Add-linux-oe-g-platform.patch ). 

Maybe those headers define that macro?

alistair23, 2019-10-24:
The macro is defined from glibc `bits/floatn.h` so it's not surprising that it's being included, the problem is that the cross compile seems to be using the host.

alistair23, 2019-10-26:
I can reproduce this with the qemuarm64 machine as well, when I run `MACHINE=qemuarm64 bitbake core-image-plasma-mobile` I get the same failure.

```
DITRO_FEATURES += "wayland x11 kde opengl pam polkit xattr ptest ipv4 ipv6
EXTRA_IMAGE_FEATURES = "debug-tweaks x11 tools-sdk dev-pkgs package-management"
```

alistair23, 2019-11-04:
Steps to reproduce:
```
git clone git://git.openembedded.org/openembedded-core
cd openembedded-core
git clone git://git.openembedded.org/bitbake
git clone https://github.com/meta-qt5/meta-qt5.git
source oe-init-build-env
bitbake-layers add-layer "$(pwd)/oe-${BRANCH}/meta-openembedded/meta-oe"
MACHINE=qemuarm64 bitbake qttools
```

alistair23, 2020-02-05:
This seems to work again, not sure if it was fixed in meta-qt5 or if it was some host issue (Arch) that has been resolved.

closed 2020-02-05 by alistair23