meta-qt5/meta-qt5 #253: Recipe request: PySide2

damdinsharav, 2019-11-05:
As title says, is it possible to add PySide2 recipe

snke, 2020-04-27:
That would be great. It should be possible from a technical point of view, seeing that there are packages for Raspbian for example. Not sure how hard it would be to create a Yocto recipe for it. 

Surely would add some more exposure to PySide2 and would make embedded developers try it out.

problame, 2020-04-27:
Funnily enough, I just spent the day attempting to create a PySide2 recipe.

PySide2 depends on shiboken2 (comes with the same repo), and apparently cross-compiling shiboken2 is

> not easily possible -> build system needs major work

(source: https://wiki.qt.io/Qt_for_Python_Development_Notes_2019 )

See also https://bugreports.qt.io/browse/PYSIDE-1033

The problem I'm stuck with is that my recipe (cross) builds shiboken2 for ARM, but then cmake attempts to invoke shiboken2, resulting in 

```
/bin/sh: shiboken2: command not found
```

Problem is: I'm also very new to Yocto, so chances are it's rather my Yocto skills than the PySide2 / shiboken2 build system that's the problem here :/

@snke would you like to collaborate  / have more experience with Yocto packaging?

problame, 2020-04-27:
The actual tracking issue seems to be https://bugreports.qt.io/browse/PYSIDE-802

snke, 2020-04-27:
> Funnily enough, I just spent the day attempting to create a PySide2 recipe.

Awesome!
 
> PySide2 depends on shiboken2 (comes with the same repo), and apparently cross-compiling shiboken2 is
>
>
> (source: https://wiki.qt.io/Qt_for_Python_Development_Notes_2019 )
>
> See also [...]

Without having seen your recipe I would at least guess you'd need to add some `-native` to some DEPENDS somewhere.

> Problem is: I'm also very new to Yocto, so chances are it's rather my Yocto skills than the PySide2 / shiboken2 build system that's the problem here :/
> 
> @snke would you like to collaborate / have more experience with Yocto packaging?

I could take a stab at it, sure. Not a real pro, but a bit of experience.
Can you upload your current recipe somewhere? (A meta-qt5 fork or whatever)
No guarantees though ;)

problame, 2020-04-28:
@snke I created a draft pull request with my current work and sent you an invite to the forked repository. Feel free to push on top of the pyside2 branch, just don't force-push

snke, 2020-05-11:
Hi @problame, sorry I was quite busy and I slept on accepting the invite, can you resend it please and I hope I'll have time to look into your changes in the next couple of days. Thank you!

problame, 2020-05-11:
@snke FWIW we have given up on porting pyside2 for now (we chose C++). Given the comments on the PySide upstream issue, I figure it won't be long until they fix their build system to support cross-compilation without the current pains.

problame, 2020-05-11:
( = feel free to fork https://github.com/49nord/meta-qt5/tree/pyside2 yourself and submit a new PR to this repo once/if you get it working)

snke, 2020-05-14:
> @snke FWIW we have given up on porting pyside2 for now (we chose C++). Given the comments on the PySide upstream issue, I figure it won't be long until they fix their build system to support cross-compilation without the current pains.

Hi @problame,

I see, thanks for the update! We also played around with PyQt for now, as it was already available as a recipe and porting to pyside2/Qt for Python shouldn't be that much of a hassle probably.

You're right, waiting for upstream to facilitate cross compiling will make this task a whole lot easier.

Thank you for work, it still should be useful for anybody else venturing further with this!

hyyoxhk, 2023-02-16:
Does anyone else care about this issue, I want to port pyside2-5.15 version on branch kirkstone too, but i don't have any idea

hyyoxhk, 2023-02-16:
Does anyone else follow with this issue? I want to port pyside2-5.15 on the branch kirkstone too, but i don't have any idea

hyyoxhk, 2023-02-16:
Does anyone else follow with this issue? I want to port pyside2-5.15 on the branch kirkstone too, but i have no idea

```
boken2Targets.cmake:87 (message):
|   The imported target "Shiboken2::shiboken2" references the file
| 
|      "/usr/bin/shiboken2"
| 
|   but this file does not exist.  Possible reasons include:
| 
|   * The file was deleted, renamed, or moved to another location.
| 
|   * An install or uninstall procedure did not complete successfully.
| 
|   * The installation package was faulty and contained
```

mhouse1, 2023-06-13:
the pyside/shiboken cross compile issue was fixed in 04 Feb, 22.
source: https://codereview.qt-project.org/c/pyside/pyside-setup/+/374068?tab=comments
of note: you'll need to use pyside6.3+ and follow the setup.py instructions