meta-qt5/meta-qt5 #266: Missing python-native

roykollensvendsen, 2019-12-27:
| sh: python: command not found
| Project ERROR: Building QtQml requires Python.
| WARNING: /yocto_build/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/qtdeclarative/5.13.2+gitAUTOINC+4080025fed-r0/temp/run.do_configure.19880:1 exit 1 from 'exit 1'

I had to add the following bbappend to make qtdeclarative to build:

cat qtdeclarative_git.bbappend 

DEPENDS += "python-native"
EXTRANATIVEPATH += "python-native"

shr-project, 2019-12-27:
Do you have this commit?
https://github.com/meta-qt5/meta-qt5/commit/f264e5dd1b589a1cf2d3904bd8c88605de43c5f6

roykollensvendsen, 2019-12-27:
No. I cloned https://code.qt.io/cgit/yocto/meta-qt5.git/ and found the same problem on dev and 5.13.2

shr-project, 2019-12-27:
Then you need to report the problem there, this github repo isn't the same as what is on code.qt.io.

closed 2019-12-27 by shr-project