meta-qt5/meta-qt5 #284: Please update wiki with information about sdk + qt5 + cmake

deblauwetom, 2020-02-09:
It is very difficult to find information on this layer and how to use it properly with an SDK.
My application compiles already since 4 days in yocto as a package. However, getting an SDK that compiles the very same application with the SDK proves to be a very difficult task.

I only need qtbase and use the cmake compilation target for my application. I am using the yocto warrior release. 
What are the steps needed? I only want a minimal image, so I don't want any X or wayland related packages, I don't want a GUI.

Thanks for any updates in this regard!

ydirson, 2020-02-11:
Yes, SDK information is sorely outdated.

We managed to add qt5 support to standard-sdk build by using `inherit populate_sdk_qt5` in
our image recipe.

OTOH, there is a bit problem with extended sdk:

- through the following chain, this is pulling `nativesdk-qemu-helper` into the sdk rootfs:
  - `inherit populate_sdk_qt5`
  - `TOOLCHAIN_HOST_TASK_append = " nativesdk-packagegroup-qt5-toolchain-host"`
  - `RDEPENDS_${PN} += "nativesdk-packagegroup-sdk-host"`
  - `RDEPENDS_${PN} += "nativesdk-qemu-helper"`

- `populate_sdk_ext.bbclass` has an `install_tools()` function, executed through `SDK_POSTPROCESS_COMMAND_prepend_task-populate-sdk-ext`, which wants to create symlinks for, among other tools, `oe-find-native-sysroot runqemu*` ... which `nativesdk-qemu-helper` already *copied* in place

This results in a failure looking like:
```
NOTE: Executing install_tools ...
DEBUG: Executing shell function install_tools
Traceback (most recent call last):
  File "<...>/poky/scripts/lnr", line 21, in <module>
    os.symlink(target, linkname)
FileExistsError: [Errno 17] File exists: '../../../../layers/poky/scripts/oe-find-native-sysroot' -> '<...>/sdk-ext/image/<...>/sysroots/x86_64-pokysdk-linux/usr/bin/oe-find-native-sysroot'
WARNING: exit code 1 from a shell command.
DEBUG: Python function do_populate_sdk_ext finished
ERROR: Function failed: install_tools (log file is located at <...>/log.do_populate_sdk_ext.15806)
```

It does look like a conflict between 2 poky-provided files, but at least now it starts to get documented somewhere :)

deblauwetom, 2020-02-11:
Thanks for the information! I'm only using the SDK for now.
After some searching and puzzling, I found indeed out a day ago that I need to make my own image file. 
So in my own layer I copied core-image-minimal to the file recipes-core/images/my-image.bb and added this in the file:
```
inherit populate_sdk_qt5_base

TOOLCHAIN_HOST_TASK_append = " nativesdk-qttools-tools"
```
So this seems to be enough to make the SDK contain qmake and still have the other existing tools like cmake too.

ydirson, 2020-02-11:
For completeness, just found that the sdk-ext error I get has been fixed in thud, by commit
"populate_sdk_ext.bbclass: avoid populating tools twice" :)

RobertBerger, 2020-09-25:
sdk_ext seems to build now (dunfell), but the extensible SDK is unusable: https://github.com/meta-qt5/meta-qt5/issues/356