meta-qt5/meta-qt5 #297: python2 in qtwebengine in qt5.15

chbae, 2020-03-31:
python2 is officially deprecated but qtwebengine in qt5.15 is required.
I would like to remove meta-python2 layer for qtwebengine.

`
inherit ${@bb.utils.contains("BBFILE_COLLECTIONS", "meta-python2", "pythonnative", "", d)}

python() {
    if 'meta-python2' not in d.getVar('BBFILE_COLLECTIONS').split():
        raise bb.parse.SkipRecipe('Requires meta-python2 to be present.')
}`

kraag-gorim, 2020-03-31:
This restriction is also present in v5.14.2 of qt.

shr-project, 2020-03-31:
It doesn't work with python3native (at least last time I've checked), just removing this restriction isn't good enough.

cpboyd, 2020-05-05:
This has been an open issue on Chromium, which qtwebengine is based upon:
https://bugs.chromium.org/p/chromium/issues/detail?id=61357
https://bugs.chromium.org/p/chromium/issues/detail?id=942720

I'd say this issue is probably not really going anywhere until it's resolved on Chromium itself.
(You could always try to submit PRs to resolve the issues blocking Chromium 942720)

Until then, there's always this python2 layer:
https://git.openembedded.org/meta-python2/

sarnold, 2023-05-01:
This is apparently still true as of today?  In this case it happens with qt4-creator package; has anything changed?

shr-project, 2023-05-01:
It changed in newer chromium, but meta-qt5 is stuck with old chromium where python3 migration is far from simple.

einarjon, 2024-06-19:
Just wanted to point out that the 87-based branch that chromium is using has added a [patch that drops the Python2 requirement](https://code.qt.io/cgit/qt/qtwebengine-chromium.git/commit/?h=87-based&id=68302c9ea158fbc83cd28570a0560e5a892b45e8).

So maybe someone wants to bump the SRCREV_chromium and clean this up?

shr-project, 2024-06-19:
Just wanted to point out that patches are welcome from someone still using this (we have switched to meta-qt6 long time ago and I'm not interested to spend even more of my free time on this layer - https://github.com/meta-qt5/meta-qt5/issues/349#issuecomment-691077189).