meta-qt5/meta-qt5 #302: qwt-qt5 building failed: filed has imcomplete type

taskx6004, 2020-04-09:
Machine: x86-64/tune=core2; raspberrypi3/4-32/64

Clang enabled or disable both fail for same reason

Did something missing in the qwt-qt5 recipe?

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j3 MAKEFLAGS=-j3 OE_QMAKE_CC=x86_64-poky-linux-gcc  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot OE_QMAKE_CXX=x86_64-poky-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0=/usr/src/debug/qwt-qt5/6.1.4-r0                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0=/usr/src/debug/qwt-qt5/6.1.4-r0                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot-native=  OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0=/usr/src/debug/qwt-qt5/6.1.4-r0                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0=/usr/src/debug/qwt-qt5/6.1.4-r0                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden OE_QMAKE_LINK=x86_64-poky-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now OE_QMAKE_AR=x86_64-poky-linux-gcc-ar OE_QMAKE_STRIP=echo OE_QMAKE_INCDIR_QT=/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot//usr/include
| cd src/ && ( test -e Makefile || /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/qwt-6.1.4/src/src.pro QWT_CONFIG+=QwtPkgConfig QWT_CONFIG+=QwtExamples ) && make -f Makefile
| make[1]: Entering directory '/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/build/src'
| compiling ../../qwt-6.1.4/src/qwt_abstract_scale_draw.cpp
| compiling ../../qwt-6.1.4/src/qwt_clipper.cpp
| compiling ../../qwt-6.1.4/src/qwt_color_map.cpp
| ../../qwt-6.1.4/src/qwt_clipper.cpp: In member function 'QVector<QwtInterval> QwtCircleClipper::clipCircle(const QPointF&, double) const':
| ../../qwt-6.1.4/src/qwt_clipper.cpp:354:23: warning: 'void qSort(Container&) [with Container = QList<double>]' is deprecated: Use std::sort [-Wdeprecated-declarations]
|   354 |         qSort( angles );
|       |                       ^
| In file included from /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtCore/qvarlengtharray.h:45,
|                  from ../../recipe-sysroot/usr/include/QtCore/qmetatype.h:48,
|                  from ../../qwt-6.1.4/src/qwt_interval.h:14,
|                  from ../../qwt-6.1.4/src/qwt_clipper.h:14,
|                  from ../../qwt-6.1.4/src/qwt_clipper.cpp:10:
| /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtCore/qalgorithms.h:188:46: note: declared here
|   188 | QT_DEPRECATED_X("Use std::sort") inline void qSort(Container &c)
|       |                                              ^~~~~
| ../../qwt-6.1.4/src/qwt_clipper.cpp:354:23: warning: 'void qSort(Container&) [with Container = QList<double>]' is deprecated: Use std::sort [-Wdeprecated-declarations]
|   354 |         qSort( angles );
|       |                       ^
| In file included from /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtCore/qvarlengtharray.h:45,
|                  from ../../recipe-sysroot/usr/include/QtCore/qmetatype.h:48,
|                  from ../../qwt-6.1.4/src/qwt_interval.h:14,
|                  from ../../qwt-6.1.4/src/qwt_clipper.h:14,
|                  from ../../qwt-6.1.4/src/qwt_clipper.cpp:10:
| /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtCore/qalgorithms.h:188:46: note: declared here
|   188 | QT_DEPRECATED_X("Use std::sort") inline void qSort(Container &c)
|       |                                              ^~~~~
| compiling ../../qwt-6.1.4/src/qwt_column_symbol.cpp
| compiling ../../qwt-6.1.4/src/qwt_date.cpp
| compiling ../../qwt-6.1.4/src/qwt_date_scale_draw.cpp
| ../../qwt-6.1.4/src/qwt_date.cpp: In static member function 'static int QwtDate::utcOffset(const QDateTime&)':
| ../../qwt-6.1.4/src/qwt_date.cpp:719:42: warning: 'int QDateTime::utcOffset() const' is deprecated: Use offsetFromUtc() instead [-Wdeprecated-declarations]
|   719 |             seconds = dateTime.utcOffset();
|       |                                          ^
| In file included from ../../qwt-6.1.4/src/qwt_date.h:14,
|                  from ../../qwt-6.1.4/src/qwt_date.cpp:10:
| ../../recipe-sysroot/usr/include/QtCore/qdatetime.h:375:56: note: declared here
|   375 |     QT_DEPRECATED_X("Use offsetFromUtc() instead") int utcOffset() const;
|       |                                                        ^~~~~~~~~
| ../../qwt-6.1.4/src/qwt_date_scale_draw.cpp: In member function 'QDateTime QwtDateScaleDraw::toDateTime(double) const':
| ../../qwt-6.1.4/src/qwt_date_scale_draw.cpp:274:44: warning: 'void QDateTime::setUtcOffset(int)' is deprecated: Use setOffsetFromUtc() instead [-Wdeprecated-declarations]
|   274 |         dt.setUtcOffset( d_data->utcOffset );
|       |                                            ^
| In file included from ../../qwt-6.1.4/src/qwt_date.h:14,
|                  from ../../qwt-6.1.4/src/qwt_date_scale_draw.h:15,
|                  from ../../qwt-6.1.4/src/qwt_date_scale_draw.cpp:10:
| ../../recipe-sysroot/usr/include/QtCore/qdatetime.h:374:60: note: declared here
|   374 |     QT_DEPRECATED_X("Use setOffsetFromUtc() instead") void setUtcOffset(int seconds);
|       |                                                            ^~~~~~~~~~~~
| compiling ../../qwt-6.1.4/src/qwt_date_scale_engine.cpp
| compiling ../../qwt-6.1.4/src/qwt_dyngrid_layout.cpp
| compiling ../../qwt-6.1.4/src/qwt_event_pattern.cpp
| ../../qwt-6.1.4/src/qwt_dyngrid_layout.cpp: In member function 'void QwtDynGridLayout::init()':
| ../../qwt-6.1.4/src/qwt_dyngrid_layout.cpp:85:25: warning: 'constexpr QFlags<T>::QFlags(QFlags<T>::Zero) [with Enum = Qt::Orientation; QFlags<T>::Zero = int QFlags<Qt::Orientation>::Private::*]' is deprecated: Use default constructor instead [-Wdeprecated-declarations]
|    85 |     d_data->expanding = 0;
|       |                         ^
| In file included from ../../recipe-sysroot/usr/include/QtCore/qglobal.h:1302,
|                  from ../../qwt-6.1.4/src/qwt_global.h:13,
|                  from ../../qwt-6.1.4/src/qwt_dyngrid_layout.h:13,
[... 169 lines cut ...]
| compiling ../../qwt-6.1.4/src/qwt_graphic.cpp
| compiling ../../qwt-6.1.4/src/qwt_interval.cpp
| compiling ../../qwt-6.1.4/src/qwt_interval_symbol.cpp
| In file included from ../../qwt-6.1.4/src/qwt_graphic.cpp:11:
| ../../qwt-6.1.4/src/qwt_painter_command.h:85:22: error: field 'clipPath' has incomplete type 'QPainterPath'
|    85 |         QPainterPath clipPath;
|       |                      ^~~~~~~~
| In file included from /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtGui/qtransform.h:43,
[... cut ...]
| /home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/recipe-sysroot/usr/include/QtGui/qmatrix.h:54:7: note: forward declaration of 'class QPainterPath'
|    54 | class QPainterPath;
|       |       ^~~~~~~~~~~~
| make[1]: *** [Makefile:1505: obj/qwt_graphic.o] Error 1
| make[1]: *** Waiting for unfinished jobs....
| make[1]: Leaving directory '/home/task03/work/poky/build/tmp/work/core2-64-poky-linux/qwt-qt5/6.1.4-r0/build/src'
| make: *** [Makefile:51: sub-src-make_first-ordered] Error 2
| WARNING: exit code 1 from a shell command.

```

shr-project, 2020-04-14:
Fixed

closed 2020-04-14 by shr-project

taskx6004, 2020-04-15:
Thanks! i will test update later this week