meta-qt5/meta-qt5 #321: REQUIRED_DISTRO_FEATURES="x11 opengl" needed for all recipes

MatthewZMD, 2020-05-22:
I was having numerous weird build failures until I realized that my DISTRO_FEATURES is missing **both** x11 and opengl. 

To avoid this kind of trouble in the future, I suggest adding `REQUIRED_DISTRO_FEATURES="x11 opengl"` to all recipes, which is also adopting similar practice from layers such as meta-xfce.

What do you guys think, I can create a PR for this.

shr-project, 2020-05-22:
It depends on qtbase PACKAGECONFIG, most people including me are using meta-qt5 without x11 in DISTRO_FEATURES, so if anything it would need to be conditional on qtbase PACKAGECONFIG which is difficult to do across all the recipes (unless adding it in some bbclass which is inherited by all qt recipes - and in theory it should be inherited by every single recipe which would depend on something from meta-qt5 as well).

MatthewZMD, 2020-05-22:
qt5-creator will always need x11, right? Because I encountered this issue when trying to build qt5-creator. 

I have a reproducible steps using poky:

1. `$ . ../poky/oe-init-build-env poky-qt5`
2. In local.conf, remove x11 from DISTRO_FEATURES because it came with x11.
    DISTRO_FEATURES_remove = "x11"
3. Add path-to-meta-qt5 to bblayers.conf
4. In terminal,
```
$ bitbake -e qt5-creator &> x11.log
$ cat x11.log | grep ^DISTRO_FEATURES=
DISTRO_FEATURES="acl alsa argp bluetooth ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc  vfat largefile opengl ptest multiarch wayland vulkan pulseaudio sysvinit gobject-introspection-data ldconfig"
$ cat x11.log | grep ^PACKAGECONFIG=
PACKAGECONFIG=""
$ bitbake qt5-creator
...
ERROR: Task (.../layers/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3564 tasks of which 3534 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  .../layers/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

A different error can be reproduced by changing `DISTRO_FEATURES_remove = "x11"` to `DISTRO_FEATURES_remove = "opengl"`:
```
$ bitbake -e qt5-creator &> opengl.log
$ cat opengl.log | grep ^DISTRO_FEATURES=
DISTRO_FEATURES="acl alsa argp bluetooth ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc x11 vfat largefile  ptest multiarch wayland vulkan pulseaudio sysvinit gobject-introspection-data ldconfig"
$ cat x11.log | grep ^PACKAGECONFIG=
PACKAGECONFIG=""
$ bitbake qt5-creator
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:00
Loaded 1410 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/egl' (but .../layers/meta-qt5/recipes-qt/qt5/qtwayland_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/kernel
  virtual/mesa
  virtual/perf
NOTE: Runtime target 'qtwayland-plugins' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['qtwayland-plugins', 'virtual/egl']NOTE: Runtime target 'packagegroup-qt5-toolchain-target' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-qt5-toolchain-target', 'qtwayland-plugins', 'virtual/egl']
ERROR: Required build target 'qt5-creator' has no buildable providers.Missing or unbuildable dependency chain was: ['qt5-creator', 'packagegroup-qt5-toolchain-target', 'qtwayland-plugins', 'virtual/egl']Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

MatthewZMD, 2020-05-26:
Here is the full log without x11 in the DISTRO_FEATURES: https://pastebin.pl/view/5b67e08a

thiagomacieira, 2020-05-26:
The log shows that configure was passed `-opengl desktop` but the OpenGL libraries are not present. That's by itself a bug in the recipe. Either pass `-no-opengl` or install the required libraries.

shr-project, 2020-05-27:
mesa provides required libraries when configured to do so. User just needs to either configure mesa correctly or set PACKAGECONFIG_GL for qtbase to whatever he wants to use. There isn't single default configuration which would suit everybody.

MatthewZMD, 2020-05-27:
> User just needs to either configure mesa correctly

Can you elaborate on how to correctly configure mesa? I'm not very familiar.

> PACKAGECONFIG_GL for qtbase to whatever he wants to use.

I added `PACKAGECONFIG_GL = "no-opengl"` to local.conf, and now seeing this error:
```
ERROR: qtwayland-5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0 do_configure: Error calling .../b/poky-qt5/tmp/work/core2-64-poky-linux/qtwayland/5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0/recipe-sysroo
t-native/usr/bin/qmake -makefile -o Makefile    QT_BUILD_PARTS-=examples QT_BUILD_PARTS-=tests   .../b/poky-qt5/tmp/work/core2-64-poky-linux/qtwayland/5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0/git
/qtwayland.pro  --   -no-feature-wayland-brcm -feature-wayland-client -no-feature-wayland-drm-egl-server-buffer -feature-wayland-egl -no-feature-wayland-libhybris-egl-server-buffer -feature-wayland-server -no-fe
ature-xcomposite-egl -no-feature-xcomposite-glx                                                                                                                                                                    
ERROR: qtwayland-5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0 do_configure: Execution of '.../b/poky-qt5/tmp/work/core2-64-poky-linux/qtwayland/5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0/temp/run.do_c
onfigure.5135' failed with exit code 1:                                                                                                                                                                            
Info: creating stash file .../b/poky-qt5/tmp/work/core2-64-poky-linux/qtwayland/5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0/build/.qmake.stash                                                        
Info: creating cache file .../b/poky-qt5/tmp/work/core2-64-poky-linux/qtwayland/5.14.99+5.15-rc1+gitAUTOINC+1a5b7938c1-r0/build/.qmake.cache                                                        
                                                                                                                                                                                                                   
Running configuration tests...                                                                                                                                                                                     
Checking for Wayland client library... yes                                                                                                                                                                         
Checking for Wayland cursor library... yes                                                                                                                                                                         
Checking for wayland-scanner... yes                                                                                                                                                                                
Checking for Wayland EGL library... yes                                                                                                                                                                            
Checking for wayland-server... yes                                                                                                                                                                                 
Done running configuration tests.                                                                                                                                                                                  
                                                                                                                                                                                                                   
Configure summary:                                                                            
Qt Wayland Drivers:                                                                                                                                                                                                EGL .................................... no                                                 
  Raspberry Pi ........................... no                       
  XComposite EGL ......................... no                                                                                                                                                                    
  XComposite GLX ......................... no                       
  DRM EGL ................................ no                                                                                                                                                                    
  libhybris EGL .......................... no                         Linux dma-buf server buffer integration . no                                                                                                                                                                     Vulkan-based server buffer integration . no                         Shm emulation server buffer integration . no                                   
Qt Wayland Client Shell Integrations:                                                                                                                                                                            
[... cut ...]
ERROR: Feature 'wayland-egl' was enabled, but the pre-condition 'features.wayland-client && features.opengl && features.egl && libs.wayland-egl && (!config.qnx || features.egl-extension-platform-wayland)' failed.                                                                                                                                                                                                                  
ERROR: Feature 'wayland-egl' was enabled, but the pre-condition 'features.wayland-server && features.opengl && features.egl && libs.wayland-egl' failed.
```

thiagomacieira, 2020-05-27:
Wayland requires OpenGL. You'll have to disable it too.

If you're going to disable EGLFS, Wayland and X11, make sure that linuxfb works for your purposes. You're likely disabling too much.

MatthewZMD, 2020-05-27:
That's more complex than I need right now, I'll close this issue for now.

closed 2020-05-27 by MatthewZMD