meta-qt5/meta-qt5 #333: Project ERROR: Test config.qtwayland_client.tests.dmabuf-server-buffer tries to use undeclared library 'drm'

outbackdingo, 2020-06-22:
series dunfell fails on nvidia tegra building core-image-x11 with

Log data follows:
| DEBUG: Executing shell function qmake5_base_preconfigure
| DEBUG: Shell function qmake5_base_preconfigure finished
| DEBUG: Executing shell function do_configure
| NOTE: qmake prevar substitution: ' QT_BUILD_PARTS+=examples QT_BUILD_PARTS-=tests '
| NOTE: qmake configure substitution: '  -no-feature-wayland-brcm -feature-wayland-client -no-feature-wayland-drm-egl-server-buffer -feature-wayland-egl -no-feature-wayland-libhybris-egl-server-buffer -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx'
| Info: creating stash file /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/build/.qmake.stash
| Info: creating cache file /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/build/.qmake.cache
| 
| Running configuration tests...
| Checking for Wayland client library... yes
| Checking for Wayland cursor library... yes
| Checking for wayland-scanner... yes
| Project ERROR: Test config.qtwayland_client.tests.dmabuf-server-buffer tries to use undeclared library 'drm'
| ERROR: Error calling /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/recipe-sysroot-native/usr/bin/qmake -makefile -o Makefile    QT_BUILD_PARTS+=examples QT_BUILD_PARTS-=tests   /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/git/qtwayland.pro  --   -no-feature-wayland-brcm -feature-wayland-client -no-feature-wayland-drm-egl-server-buffer -feature-wayland-egl -no-feature-wayland-libhybris-egl-server-buffer -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/temp/run.do_configure.65512' failed with exit code 1:
| Info: creating stash file /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/build/.qmake.stash
| Info: creating cache file /home/dingo/nvidia/xavier/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/build/.qmake.cache
| 
| Running configuration tests...
| Checking for Wayland client library... yes
| Checking for Wayland cursor library... yes
| Checking for wayland-scanner... yes
| Project ERROR: Test config.qtwayland_client.tests.dmabuf-server-buffer tries to use undeclared library 'drm'
| WARNING: exit code 1 from a shell command.

schnitzeltony, 2020-06-22:
Just a guess: Tegra sounds like a closed source graphic blob thingy. From my experience  these tend to cause trouble. So I forward this to @madisongh: Is qtwayland supposed to build for tegra/dunfell?

Background: 
@outbackdingo seems interested in building kde-plasma (in [meta-qt5-extra](https://github.com/schnitzeltony/meta-qt5-extra)). I guess that because he/she/it opened an [issue in meta-mortsgna](https://github.com/schnitzeltony/meta-mortsgna/issues/16)

madisongh, 2020-06-23:
AFAIK, Qt supports at least jetson-tx1 and jetson-tx2, although Xavier should work pretty much the same way.  Looking at their boot2qt distro, they do have [this bbappend for qtbase](https://code.qt.io/cgit/yocto/meta-boot2qt.git/tree/meta-boot2qt-distro/dynamic-layers/tegra/recipes-qt/qt5/qtbase_git.bbappend?h=5.15.0_QtAS) that might be relevant to the build error mentioned.

I think they also explicitly disable Vulkan support on the tegras, since NVIDIA's Vulkan support currently only works with X11.

pingshs, 2020-09-07:
How exactly will this problem be solved? Who can tell me?

graugans, 2020-10-20:
Any news on that topic. I ran into the exact same issue today. I do neither depend on wayland, vulkan nor x11. But somehow the dependencies got pulled in when I build the SDK for my image.