meta-qt5/meta-qt5 #337: Qt 5.15 LTS for Dunfell LTS branch

fschrempf, 2020-07-07:
Currently the Dunfell branch includes Qt 5.14 and only master has Qt 5.15.  
Yocto 3.1 (Dunfell) is marked as LTS release and so is Qt 5.15. Therefore I think it would make a lot of sense to have Qt 5.15 in the Dunfell release. Is there any chance this will happen? Is there any plan or roadmap to follow?

Thanks,
Frieder

shr-project, 2020-07-07:
I have no plans to support this.

It's possible to use meta-qt5/master with dunfell. meta-qt5/dunfell needs to stay on 5.14 Qt for people who prefers that version (for whatever reason) and creating another dunfell branch in meta-qt5 is another maintenance I don't have time for.

fschrempf, 2020-07-07:
Hi Martin, thanks for letting me know!
One more question: How are the values for `LAYERSERIES_COMPAT` assigned? Will the next release branch (Gatesgarth?) have Qt 5.15 and stay compatible with Dunfell?

shr-project, 2020-07-07:
If meta-qt5/master needs some backwards incompatible (with dunfell) changes before gatesgarth is branched, then dunfell will be dropped from LAYERSERIES_COMPAT to force people to do necessary adaptations in some integration layer, like currently for dunfell/master meta-qt5 branch if you want to use it with zeus or older (as described in https://github.com/meta-qt5/meta-qt5/commit/5e18973196e3ba71ed55cff96b323e2ea10c1523)

fschrempf, 2020-07-07:
Ok, got it. Thanks a lot for providing the details.

closed 2020-07-07 by fschrempf

otavio, 2020-07-07:
We started a compat layer (for rocko, for now) and it can easily be added to support dunfell if needed.

otavio, 2020-07-07:
https://github.com/OSSystems/meta-qt5-compat

fschrempf, 2020-07-07:
> We started a compat layer (for rocko, for now) and it can easily be added to support dunfell if needed.

Good to know, thanks!