meta-qt5/meta-qt5 #341: Can't compile qtwebengine 5.15.0 and 5.14.1

BenediktCyranka, 2020-07-21:
Hi I tried to compile the qtwebengine in the Version 5.14.1 on Thud but I get a error message.
( qtwebengine: Fix build with ninja >= 1.10.0 -> 3e759d827722de6fddc9449f1149137021f4e25e 13 Mar 2020)
Gcc 8.2.0

Then I update my yocto to dunfell and tried it again. But it won't compile and I get the same error message: 
(python3-pyqt5: Update to 5.15 version  -> 1650757f4182435a63985f73e477ed80927f0eac 07 Jul 2020)
Gcc 9.3.0

`[17666/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/code-comments.o
[17667/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/code-desc.o
[17668/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/builtins.o
FAILED: v8_snapshot/obj/v8/v8_base_without_compiler/builtins.o 
/workdisk1/Bene/gen4hmi/build/tmp/hosttools/g++ -MMD -MF v8_snapshot/obj/v8/v8_base_without_compiler/builtins.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -DTOOLKIT_QT -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64 -DENABLE_MINOR_MC -DV8_INTL_SUPPORT -DV8_CONCURRENT_MARKING -DV8_ENABLE_LAZY_SOURCE_POSITIONS -DV8_EMBEDDED_BUILTINS -DV8_WIN64_UNWINDING_INFO -DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH -DV8_COMPRESS_POINTERS -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_DEPRECATION_WARNINGS -DV8_TARGET_ARCH_ARM64 -DV8_HAVE_TARGET_OS -DV8_TARGET_OS_LINUX -DDISABLE_UNTRUSTED_CODE_MITIGATIONS -DV8_COMPRESS_POINTERS -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_DEPRECATION_WARNINGS -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DUCHAR_TYPE=uint16_t -DV8_COMPRESS_POINTERS -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_DEPRECATION_WARNINGS -Iv8_snapshot/gen -I../../../../git/src/3rdparty/chromium -I../../../../git/src/3rdparty/chromium/v8 -Iv8_snapshot/gen/v8 -I../../../../git/src/3rdparty/chromium/third_party/icu/source/common -I../../../../git/src/3rdparty/chromium/third_party/icu/source/i18n -I../../../../git/src/3rdparty/chromium/v8/include -I../../../../git/src/3rdparty/chromium/third_party/zlib -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -m64 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-packed-not-aligned -Wno-dangling-else -Wno-missing-field-initializers -Wno-unused-parameter -fno-omit-frame-pointer -fvisibility=hidden -Wno-strict-overflow -Wno-return-type -O3 -fno-ident -fdata-sections -ffunction-sections -g0 -std=gnu++14 -Wno-narrowing -Wno-class-memaccess -Wno-attributes -Wno-class-memaccess -Wno-subobject-linkage -Wno-invalid-offsetof -Wno-return-type -Wno-deprecated-copy -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -c ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc -o v8_snapshot/obj/v8/v8_base_without_compiler/builtins.o
../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc: In constructor ‘v8::internal::{anonymous}::OffHeapTrampolineGenerator::OffHeapTrampolineGenerator(v8::internal::Isolate*)’:
../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:313:60: error: use of deleted function ‘v8::internal::MacroAssembler::MacroAssembler(v8::internal::Isolate*, v8::internal::CodeObjectRequired, std::unique_ptr<v8::internal::AssemblerBuffer>)’
               ExternalAssemblerBuffer(buffer_, kBufferSize)) {}
                                                            ^
In file included from ../../../../git/src/3rdparty/chromium/v8/src/codegen/macro-assembler.h:39:0,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:11:
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/macro-assembler-arm64.h:1373:25: note: ‘v8::internal::MacroAssembler::MacroAssembler(v8::internal::Isolate*, v8::internal::CodeObjectRequired, std::unique_ptr<v8::internal::AssemblerBuffer>)’ is implicitly deleted because the default definition would be ill-formed:
   using TurboAssembler::TurboAssembler;
                         ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/macro-assembler-arm64.h:1373:25: error: use of deleted function ‘v8::internal::TurboAssembler::TurboAssembler(v8::internal::Isolate*, v8::internal::CodeObjectRequired, std::unique_ptr<v8::internal::AssemblerBuffer>)’
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/macro-assembler-arm64.h:162:29: note: ‘v8::internal::TurboAssembler::TurboAssembler(v8::internal::Isolate*, v8::internal::CodeObjectRequired, std::unique_ptr<v8::internal::AssemblerBuffer>)’ is implicitly deleted because the default definition would be ill-formed:
   using TurboAssemblerBase::TurboAssemblerBase;
                             ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/macro-assembler-arm64.h:162:29: error: no matching function for call to ‘v8::internal::CPURegList::CPURegList()’
In file included from ../../../../git/src/3rdparty/chromium/v8/src/codegen/register-arch.h:16:0,
                 from ../../../../git/src/3rdparty/chromium/v8/src/codegen/interface-descriptors.h:11,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins-descriptors.h:9,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:8:
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:575:3: note: candidate: v8::internal::CPURegList::CPURegList(v8::internal::CPURegister::RegisterType, int, int, int)
   CPURegList(CPURegister::RegisterType type, int size, int first_reg,
   ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:575:3: note:   candidate expects 4 arguments, 0 provided
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:570:3: note: candidate: v8::internal::CPURegList::CPURegList(v8::internal::CPURegister::RegisterType, int, v8::internal::RegList)
   CPURegList(CPURegister::RegisterType type, int size, RegList list)
   ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:570:3: note:   candidate expects 3 arguments, 0 provided
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:562:12: note: candidate: template<class ... CPURegisters> v8::internal::CPURegList::CPURegList(v8::internal::CPURegister, CPURegisters ...)
   explicit CPURegList(CPURegister reg0, CPURegisters... regs)
            ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:562:12: note:   template argument deduction/substitution failed:
In file included from ../../../../git/src/3rdparty/chromium/v8/src/codegen/macro-assembler.h:39:0,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:11:
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/macro-assembler-arm64.h:162:29: note:   candidate expects at least 1 argument, 0 provided
   using TurboAssemblerBase::TurboAssemblerBase;
                             ^
In file included from ../../../../git/src/3rdparty/chromium/v8/src/codegen/register-arch.h:16:0,
                 from ../../../../git/src/3rdparty/chromium/v8/src/codegen/interface-descriptors.h:11,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins-descriptors.h:9,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:8:
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:559:25: note: candidate: constexpr v8::internal::CPURegList::CPURegList(const v8::internal::CPURegList&)
 class V8_EXPORT_PRIVATE CPURegList {
                         ^
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:559:25: note:   candidate expects 1 argument, 0 provided
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:559:25: note: candidate: constexpr v8::internal::CPURegList::CPURegList(v8::internal::CPURegList&&)
../../../../git/src/3rdparty/chromium/v8/src/codegen/arm64/register-arm64.h:559:25: note:   candidate expects 1 argument, 0 provided
In file included from ../../../../git/src/3rdparty/chromium/v8/src/codegen/macro-assembler.h:39:0,
                 from ../../../../git/src/3rdparty/chromium/v8/src/builtins/builtins.cc:11:
[... cut ...]
[17671/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/builtins-intl.o
[17672/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/assembler.o
[17673/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/code-factory.o
[17674/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/compilation-cache.o
[17675/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/constant-pool.o
[17676/18397] CXX v8_snapshot/obj/v8/v8_base_without_compiler/assert-scope.o`

Is there a problem in the V8 engine?

nathan546, 2020-07-27:
Saw the same.

Adding a default constructor with no parameters helps it compile for me:

`diff --git a/chromium/v8/src/codegen/arm64/register-arm64.h b/chromium/v8/src/codegen/arm64/register-arm64.h
index 6f2c7426459..d746da1ed29 100644
--- a/chromium/v8/src/codegen/arm64/register-arm64.h
+++ b/chromium/v8/src/codegen/arm64/register-arm64.h
@@ -567,6 +567,10 @@ class V8_EXPORT_PRIVATE CPURegList {
     DCHECK(is_valid());
   }
 
+  CPURegList() {
+    DCHECK(is_valid());
+  }
+
   CPURegList(CPURegister::RegisterType type, int size, RegList list)
       : list_(list), size_(size), type_(type) {
     DCHECK(is_valid());
`

BenediktCyranka, 2020-07-28:
Thanks a lot for your support ;) 
It worked.

closed 2020-07-28 by BenediktCyranka

muse117, 2022-05-17:
> Saw the same.
>
> Adding a default constructor with no parameters helps it compile for me:
>
> ```
> diff --git a/chromium/v8/src/codegen/arm64/register-arm64.h [...]

Where did this solution come from?