meta-qt5/meta-qt5 #346: Drop Down Menu is not shown

BenediktCyranka, 2020-08-13:
Hey everyboy,

I have some problems with the qtwebengine in this version of the meta-qt5:
1650757f4182435a63985f73e477ed80927f0eac

If I click on some html dropdown menus the menu is not popping out. 
I can reproduce it with this website: [Link to webpage](https://greensock.com/js/speed.html)

On the qtwebengine git I found this commit that handles some problems with the dropdown menu. But the qtwebengine version thats in the meta-qt5 is newer that this one.

[Link to Commit](https://code.qt.io/cgit/qt/qtwebengine.git/commit/?h=5.15&id=1390979a6f0a8469642bc294e4647470575016c4)

Does anyone else have the same problem or can recreate it?

BenediktCyranka, 2020-08-17:
Hey everyone,

I found this Bug in the QT Bugreports:
[Link](https://bugreports.qt.io/browse/QTBUG-86034)

Someone else have the same problems with embbeded devices.

nikodatakamp, 2020-10-22:
I'm experiencing the same issues. Have a look: [https://bugreports.qt.io/browse/QTBUG-87281](url)