meta-qt5/meta-qt5 #348: Qt 5.15 build error [ sub-qmetatype-make_first]  [sub-auto-make_first]

buizv, 2020-08-19:
Hello,
I am trying to bitbake Qt5.15 on IMX Yocto 3.0.0 (Zeus) but there are some make error I can not understand
please help me why this happend
thank you so much

/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/9.2.0/ld: /usr/src/debug/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/tests/auto/corelib/kernel/qmetatype/tst_qmetatype.cpp:2556: undefined reference to `CustomConvertibleType::s_value'
/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/9.2.0/ld: /usr/src/debug/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/tests/auto/corelib/kernel/qmetatype/tst_qmetatype.cpp:2556: undefined reference to `MessageHandlerCustom::expectedMessage'
make[5]: Entering directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/gui/image/qicon'
make[5]: Nothing to be done for 'first'.
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/gui/image/qicon'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:108: tst_qmetatype] Error 1
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/corelib/kernel/qmetatype'
make[4]: *** [Makefile:317: sub-qmetatype-make_first] Error 2
make[4]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/corelib/kernel'
cd qiconhighdpi/ && ( test -e Makefile || /home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/tests/auto/gui/image/qiconhighdpi/qiconhighdpi.pro -qtconf /home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/bin/qt.conf ) && make -f Makefile 
make[3]: *** [Makefile:60: sub-kernel-make_first] Error 2
make[3]: *** Waiting for unfinished jobs....

make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/watchdog'
make[5]: Entering directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/keyboard'
make[5]: Nothing to be done for 'first'.
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/keyboard'
make[5]: Entering directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/xunit'
make[5]: Nothing to be done for 'first'.
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/xunit'
make[5]: Entering directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/mouse'
make[5]: Nothing to be done for 'first'.
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/mouse'
make[5]: Entering directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/test'
make[5]: Nothing to be done for 'first'.
make[5]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests/test'
make[4]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib/selftests'
make[3]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto/testlib'
make[2]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests/auto'
make[1]: *** [Makefile:48: sub-auto-make_first] Error 2
make[1]: Leaving directory '/home/ubuntu/Workspace/imx-yocto-bsp/build-imx-image-qt515/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/qtbase/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/tests'
make: *** [Makefile:101: sub-tests-make_first] Error 2
WARNING: exit code 1 from a shell command.

buizv, 2020-08-20:
maybe i found the problem where is, but dont understand why
when i build yocto with meta-qt5 on Ubuntu 20.04 something wrong or uncompatible cause this error
I try on Ubuntu 18.04.4 again, everything is good

buizv, 2020-08-20:
Update: I think Ubuntu version is not cause. I relized that I need to install some build tools extra from apt-get, but not understand exactly which is. 

sudo apt-get install bison g++-multilib git gperf libxml2-utils make python-networkx zip
sudo apt-get install flex curl libncurses5-dev libssl-dev zlib1g-dev gawk minicom
sudo apt-get install exfat-fuse exfat-utils device-tree-compiler liblz4-tool

afinfante, 2022-01-21:
Hello @buizv. I have a similar [problem](https://github.com/meta-qt5/meta-qt5/issues/443)