meta-qt5/meta-qt5 #349: Qt 6 plans

shr-project, 2020-08-19:
With Qt 6 Feature Freeze close (last meeting minutes said 31st August 2020), we should discuss how to support it in OE.

I personally don't have any plans to work on it (lack of free time and motivation). I'll continue to support what is here already in my limited spare time.

This layer is called meta-qt5, adding Qt 6 here doesn't make much sense, it probably won't be able to reuse much from qt5 bbclasses and configuration, so whoever volunteers to work on Qt 6 should probably start by creating new meta-qt6 org and meta-qt6 repository on github.

If you plan to work on it, then please leave an comment here, so that other interested people know where to look.

otavio, 2020-08-19:
I am not too motivated in work on it but I can help whoever intends to work on it.

schnitzeltony, 2020-08-19:
I prefer a new layer either.

Currently: motivation high / resources low. So if somebody stand up I would throw meta-qt5-extra on it.

Steffko, 2020-08-24:
I am motivated to help and support the integration of Qt6 into a new meta-qt6 layer, but isn't it useful to discuss the parts of the meta-qt5 layer, which are reusable for a possible new meta-qt6 layer?

kanavin, 2020-09-08:
Qt Company is actively working on meta-qt6:
https://code.qt.io/cgit/yocto/meta-qt6.git/log/

https://bugreports.qt.io/browse/QTBUG-81928

As long as they're doing a good job, I guess we could let them take over?

@sapiippo what's your take? Any comments?

otavio, 2020-09-08:
The Qt policy for the layer changes is not well aligned with the way we usually maintain Yocto Project layers. Some of the changes they tried to do in the past we denied and I always prefer when this kind of key component is maintained in a more neutral way.

Maintain Qt is time-consuming and Martin has done a great job. If Qt is willing to do it, it is good but it'd be nice if it respects the good practices. I am a little concerned since Qt last tentative of closing source and line.

Steffko, 2020-09-08:
I agree with Otavio here. My experience with the applications, dependent on
Qt5, were that meta-qt5 from the community was a lot more stable and easier
to maintain then the Qt5 layer from Qt.

But it's just the view from a application developer here.

A fork/doubling is never a good way, but maybe it makes sense, to maintain
such an important key framework in the community as well.

If Qt has a monopoly on the layer here, it could implicitly affect the
mainline layer quality.

schnitzeltony, 2020-09-08:
Just reading https://bugreports.qt.io/browse/QTBUG-81928 does not motivate me exactly to contribute there...

sapiippo, 2020-09-09:
The work with meta-qt6 layer has been going on for a while now in the Qt company. It's been very unstable up until now since the work on Qt side for the cmake transition has been ongoing as well. With the Qt 6 feature freeze now in place, the Qt code is starting to get stable and I feel the meta-qt6 is also getting into good shape.

The layer was started in internal repo, but was always meant to be opened up. Only question was where it should be and how the maintenance and contributions should look like. The layer was made visible to public last week in preparation for this very discussion.

Currently the layer is open, MIT licensed and accepting contributions in the same manner as the open source Qt using the Gerrit codereview: https://codereview.qt-project.org/q/project:yocto/meta-qt6. The layer is also using the Qt CI and does build tests for few qemu targets with different configurations, CI is run for all commits before they are merged in.

shr-project, 2020-09-11:
I'm glad that Qt company works on meta-qt6.

And I don't agree with
 > We have done most of the Qt module update changes while making sure meta-qt5 works with latest yocto has been done by the opensource maintainers.

from https://bugreports.qt.io/browse/QTBUG-81928 so as @schnitzeltony I'm not really motivated to contribute to meta-qt6.

Here are some stats:
```
meta-qt5$ git log | grep ^Author | sort | uniq -c | sort -n | tail
     19 Author: Denys Dmytriyenko <denys@ti.com>
     20 Author: Eric Bénard <eric@eukrea.com>
     27 Author: Samuli Piippo <samuli.piippo@theqtcompany.com>
     29 Author: Andreas Müller <schnitzeltony@gmail.com>
     37 Author: Andreas Müller <schnitzeltony@googlemail.com>
     51 Author: Samuli Piippo <samuli.piippo@qt.io>
     53 Author: Otavio Salvador <otavio@ossystems.com.br>
     58 Author: Jonathan Liu <net147@gmail.com>
    121 Author: Khem Raj <raj.khem@gmail.com>
    417 Author: Martin Jansa <Martin.Jansa@gmail.com>
```

otavio, 2020-09-11:
Neither am I as Qt seems to not see value our contributions and then I see no value in contributing to the project.

sapiippo, 2020-09-11:
The comment was not meant to belittle your contributions and I'm sorry if it gave you that impression. I know that Jansa especially has done a lot of work with the layer and that has allowed us to focus on testing the latest Qt versions instead.

If you'd take the same stats from the our mirror, where we do our daily work, you'd see that we have a lot of commits that keep the latest version of Qt available there. As those are unreleased and sometimes unstable, we haven't seen the need to push those here. Instead we have contributed the released versions when ever Jansa hasn't done the same work.

> I see no value in contributing to the project.

I'd see the value in having a good meta layer regardless of _where_ that layer is maintained.