meta-qt5/meta-qt5 #354: Unable to load .../v8/snapshot_toolchain.gni

ChenQi1989, 2020-09-08:
Error message is as below:

| ERROR at //third_party/icu/BUILD.gn:10:1: Unable to load "/path/to/build/tmp-glibc/work/core2-32-wrs-linux/
qtwebengine/5.15.0+gitAUTOINC+f526855509_7b2f027ea8-r0/git/src/3rdparty/chromium/v8/snapshot_toolchain.gni".
| import("//v8/snapshot_toolchain.gni")
| ^-----------------------------------
| See //url/BUILD.gn:97:15: which caused the file to be included.
|     deps += [ "//third_party/icu" ]
|               ^------------------
| Project ERROR: GN run error!
| Makefile:82: recipe for target 'sub-gn_run-pro-make_first' failed

I found the snapshot_toolchain is actually under v8/gni/ instead of v8/, so I changed it locally.
But another do_compile failure follows.
/usr/include/c++/7/bits/stl_algobase.h:59:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.

The above could be solved by installing g++-multilib, but it's not a required host package for yocto.

I'm wondering whether you have encountered the similar problem and how you solves it?

schnitzeltony, 2020-09-08:
I think this is same as fixed by https://github.com/meta-qt5/meta-qt5/pull/353 - so I am not the only one with problems on this....

ChenQi1989, 2020-09-08:
@schnitzeltony Glad to know there's already a fix.

Anyway, did you meet the following error after the fix?
/usr/include/c++/7/bits/stl_algobase.h:59:10: fatal error: bits/c++config.h: No such file or directory

Or you have g++-multilib installed on host?

schnitzeltony, 2020-09-08:
Yes something like that mentioned in https://github.com/meta-qt5/meta-qt5/blob/master/README.md

ChenQi1989, 2020-09-08:
@schnitzeltony THANKS. Good to know the info.