meta-qt5/meta-qt5 #376: python3-pyqt5 fetch fails

sakoman, 2020-12-11:
Looks like upstream version has been bumped to 5.15.3 and the old version removed

sakoman, 2020-12-11:
Neglected to indicate I'm using the dunfell branch!

bernardoaraujor, 2020-12-13:
same problem here!
trying to build at dunfell

this URL seems broken:
https://www.riverbankcomputing.com/static/Downloads/PyQt5/5.13.2/PyQt5-5.13.2.tar.gz

bernardoaraujor, 2020-12-14:
cc @otavio

sakoman, 2021-01-12:
Any updates on when this might be fixed?

bernardoaraujor, 2021-01-12:
> Any updates on when this might be fixed?

PR pending: https://github.com/meta-qt5/meta-qt5/pull/378

shr-project, 2021-01-22:
> > Any updates on when this might be fixed?
> 
> PR pending: #378

it's pending but blocked by sip in meta-oe based on your comment there:

> PyQt5 v5.15.1 depends on sip3 v4.19.23 or later.
> However meta-oe Dunfell provides sip3 v4.19.19.

> in order for this PR to be effective, meta-oe Dunfell would need to bump sip3 as well.

Anyone serious about OE should have own PREMIRROR with this archive already there.

bernardoaraujor, 2021-01-22:
> Anyone serious about OE should have own PREMIRROR with this archive already there.

@shr-project could you elaborate?

sakoman, 2021-01-22:
I'm surprised at this comment.  What about people new to meta-qt5 who want to use the dunfell LTS release?  Their first experience with meta-qt5 will not be good as a result of this issue. How are they to know that they must set up a PREMIRROR (if they even know what that is) and where are they to find the missing tarball?

shr-project, 2021-01-22:
> I'm surprised at this comment. What about people new to meta-qt5 who want to use the dunfell LTS release? Their first experience with meta-qt5 will not be good as a result of this issue. How are [...]

Yes, it's far from ideal, I wish https://www.riverbankcomputing.com/ wouldn't be removing the archives, but replacing version which currently doesn't fetch from upstream (but people might have it in their mirrors already) with a version which fetches, but doesn't build, because sip in meta-oe/dunfell isn't new enough doesn't improve this situation, especially as even this version will be eventually removed from https://www.riverbankcomputing.com/ again.

It's pity that http://sources.openembedded.org/ are no longer updated by world builds (PyQt5_gpl-5.12.1.tar.gz is the latest version available there). Maybe we can ask @kraj to check why it's no longer populated by his builds. I'll upload the tarballs to http://sources.webos-ports.org/ so that people can manually fetch it from there (and put to their mirror), but unfortunately this mirror isn't enabled by default (unlike http://sources.openembedded.org/).

kraj, 2021-01-22:
Meta-qt5 is not part of layers on yoe builds we do for meta-oe CI

shr-project, 2021-03-04:
BTW: PyQtChart-5.13.1.tar.gz also fails to fetch, will upload it like the PyQt5-5.13.2.tar.gz to http://sources.webos-ports.org/