meta-qt5/meta-qt5 #398: [GATESGARTH] qtwebengine do_compile fails when PACKAGECONFIG includes libwebp

sakoman, 2021-03-03:
I have a qtwebengine bbappend which sets PACKAGECONFIG:

PACKAGECONFIG = "ffmpeg libwebp opus libvpx webrtc zlib jsoncpp \
                   ${@bb.utils.contains('DISTRO_FEATURES', 'pulseaudio', 'pulseaudio', '', d)}"

Build fails with following error:

| ERROR at //build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
|     pkgresult = exec_script(pkg_config_script, args, "value")
|                 ^----------
| Current dir: /home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/build/src/core/release/
| Command: /home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/recipe-sysroot-native/usr/bin/python-native/python2 /home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/git/src/3rdparty/chromium/build/config/linux/pkg-config.py -p /home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/recipe-sysroot-native/usr/bin/pkg-config-native libwebp libwebpmux libwebpdemux
| Returned 1.
| stderr:
| 
| Package libwebp was not found in the pkg-config search path.
| Perhaps you should add the directory containing `libwebp.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'libwebp' found
| Package libwebpmux was not found in the pkg-config search path.
| Perhaps you should add the directory containing `libwebpmux.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'libwebpmux' found
| Package libwebpdemux was not found in the pkg-config search path.
| Perhaps you should add the directory containing `libwebpdemux.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'libwebpdemux' found
| Could not run pkg-config.
| 
| See //third_party/libwebp/BUILD.gn:420:3: whence it was called.
|   pkg_config("system_libwebp") {
|   ^-----------------------------
| See //third_party/blink/renderer/platform/BUILD.gn:170:5: which caused the file to be included.
|     "//third_party/libwebp",
|     ^----------------------
| Project ERROR: GN run error!
| make[2]: *** [Makefile:82: sub-gn_run-pro-make_first] Error 3
| make[2]: Leaving directory '/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/build/src/core'
| make[1]: *** [Makefile:79: sub-core-make_first] Error 2
| make[1]: Leaving directory '/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/build/src'
| make: *** [Makefile:49: sub-src-make_first] Error 2
| WARNING: /home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-imx-reach-linux-gnueabi/qtwebengine/5.15.2+gitAUTOINC+5537ff4437_f5a93d251c-r0/temp/run.do_compile.3902492:1 exit 1 from 'exit 1'
| 

This worked prior to the version bump to 5.15.2

shr-project, 2021-03-03:
@jasuonpe Can you please have a look?

jasuonpe, 2021-03-05:
Hi @shr-project and @sakoman, 
Package configs
PACKAGECONFIG = "ffmpeg libwebp opus libvpx webrtc zlib jsoncpp
${@bb.utils.contains('DISTRO_FEATURES', 'pulseaudio', 'pulseaudio', '', d)}"

requires native versions of libwebp and libvpx. There must be next lines in bbappend of qtwebengine 
DEPENDS += " \
    libvpx-native \
    libwebp-native \
"

Next problem is that native versions of libwebp and libvpx are not available. It seems that libwebp can append to native with bbappend file
BBCLASSEXTEND = "native"
libvpx does not be so simple. 

@sakoman you can make own bbappend files to extend to libwebp and libvpx. I think that @shr-project know much better and I how to get native support for libwebp and libvpx.

sakoman, 2021-03-07:
@jasuonpe Thanks for investigating this.  I added the BBCLASSEXTEND in a libwebp bbappend, and a DEPENDS for libwebp-native in my qtwebengine bbappend.  That seemed to be sufficient for a successful build on Ubuntu 18.04 and 20.04.  Which distro/version were you using when you found the issue with libvpx-native?

jasuonpe, 2021-03-08:
@sakoman I am using Ubuntu 20.04. It this case it might be some other issue in my environment. If you succeeded to build both then it is good. My QT 5.15 Yocto uses Zeus branches. I can see subject of that issue that your branch is Gatesgarth.