meta-qt5/meta-qt5 #436: [meta-qt5] populate_sdk fails with latest master branch

matthiasklein, 2021-11-22:
I am trying to use the meta-qt5 layer in conjunction with the current Yocto master branch.

In my image I added `inherit populate_sdk_qt5` as well as `IMAGE_INSTALL:append = " qtbase qtserialport qtserialbus"`.

The image builds fine, but when trying to build the SDK to it, **nativesdk-qtbase** aborts at do_configure with the following error:

`ERROR: Feature 'pkg-config' was enabled, but the precondition 'tests.pkg-config' failed.`

To fix the problem I created a `nativesdk-qtbase_git.bbappend` with the following content:

`DEPENDS:append = " pkgconfig-native"`

Can anyone confirm the problem?

shr-project, 2021-11-22:
It's already fixed for target qtbase by https://github.com/meta-qt5/meta-qt5/commit/0ae3cbd677b6a8b9f7571a3f3287d9ebe144fbb4 nativesdk-qtbase probably needs the same, I usually don't build sdk, so I haven't noticed. Will run a build and fix it.

closed 2021-11-22 by shr-project

matthiasklein, 2021-11-23:
Thanks for the quick reply and fix!