meta-qt5/meta-qt5 #443: [dunfell] meta-qt5 layer do_compile issue

afinfante, 2022-01-21:
I am having a problem that I detail in the following stackoverflow [post]((https://stackoverflow.com/questions/70767215/meta-qt5-layer-issue-yocto-workflow) in which I define the enviroment and attach log.

madscientist42, 2023-01-21:
Looking at the log...it seems to be compiling instead of cowpiling things, it's just REALLLY...ssssssslllllloooooowww...at doing it for the stuff in question.  Looks like something that got picked up with the latest of meta-qt5 and the version they have a tip their repo.

madscientist42, 2023-01-21:
It's worth note that there's something that keeps it from doing anything other than a sequential build  there- it's PART of why it seems so slow.  Painfully slow as you've described.  What's troubling to me is that NOBODY has helped step up and try to figure it out until I went looking for answers on the same problem.  Seriously, people?

madscientist42, 2023-01-21:
Well, then...

> DEBUG: Executing shell function do_compile
NOTE: make -j 1 MAKEFLAGS=-j 1 OE_QMAKE_CC=aarch64-poky-linux-gcc  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot OE_QMAKE_CXX=aarch64-poky-linux-g++  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0=/usr/src/debug/qtbase/5.15.7+gitAUTOINC+358aebba72-r0                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0=/usr/src/debug/qtbase/5.15.7+gitAUTOINC+358aebba72-r0                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot=                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot-native=   -fmacro-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/image=  OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0=/usr/src/debug/qtbase/5.15.7+gitAUTOINC+358aebba72-r0                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0=/usr/src/debug/qtbase/5.15.7+gitAUTOINC+358aebba72-r0                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot=                      -fdebug-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot-native=   -fmacro-prefix-map=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/image=  -fvisibility-inlines-hidden OE_QMAKE_LINK=aarch64-poky-linux-g++  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now OE_QMAKE_AR=aarch64-poky-linux-gcc-ar OE_QMAKE_OBJCOPY=aarch64-poky-linux-objcopy OE_QMAKE_STRIP=echo OE_QMAKE_INCDIR_QT=/space/trx/build/tmp/work/zynqmp-eg-poky-linux/qtbase/5.15.7+gitAUTOINC+358aebba72-r0/recipe-sysroot//usr/include

madscientist42, 2023-01-21:
It's forcing itself to -j 1, in defiance to any settings you might have to parallel compile.  Heh...  Now to dig for WHY.

It's embarrassing to have me doing this, guys.  Do you even maintain this?

afinfante, 2023-02-17:
@madscientist42 what are you trying to accelerate the build? I don't know if this will be the case, since you may already do it, but you can accelerate the development flow as detailed [here](https://docs.yoctoproject.org/dev/dev-manual/speeding-up-build.html) (in particular see the `PARALLEL_MAKE` flag)