meta-qt5/meta-qt5 #456: Qt5 Fails to build with gcc11 on Ubuntu 21.10

emz229, 2022-03-04:
when attempting to build I received various errors such as:

```
/usr/include/c++/11/limits:1818:38: error: ‘std::numeric_limits’ is not a template
|  1818 |     struct numeric_limits<long double>
``
The following patch allows me to build:

[0001-qt5-base-Fix-numeric_limits-error-when-using-GCC11.zip](https://github.com/meta-qt5/meta-qt5/files/8189121/0001-qt5-base-Fix-numeric_limits-error-when-using-GCC11.zip)
`

shr-project, 2022-03-07:
```
$ bitbake-getvar -r qtbase-native PACKAGECONFIG
```
the default is empty.

And we already have this fixed in various header files, see:
https://github.com/meta-qt5/meta-qt5/blob/cc3d1c0c9b62f2062e980562c38b11d7dbdede1c/recipes-qt/qt5/qtbase/0021-qfloat16-Include-limits-header.patch

The qfloat16.h part is now also resolved upstream, other 2 are still needed with 5.15.3-lts-lgpl.

Including whole build log might also show which part failed to build for you.

emz229, 2022-03-07:
the bitbake-getvar command is not present in the version of bitbake that we are using, we are building based off the 1.46 branch from [github.com/openembedded/bitbake](url)

dv1, 2022-04-15:
A fix for this is included in the dunfell-next branch. I also added a PR to fix [a similar problem when building qtdeclarative](https://github.com/meta-qt5/meta-qt5/pull/462).

shr-project, 2022-04-16:
I've merged both fixes to dunfell, but it's not clear what branch/version @emz229 was building.

sakoman, 2022-04-29:
Would it be possible to get these gcc11 patches in gatesgarth too?

shr-project, 2022-04-29:
@sakoman I've cherry-picked them from dunfell to gatesgarth, can you please test from your side? It's possible that it won't apply, but I don't have any build environment to test this combination.

sakoman, 2022-04-29:
@shr-project I just installed Ubuntu 22.04 on a machine here and will give gatesgarth-next a try a bit later this morning.

Unfortunately the client needs the version in gatesgarth for the next few months, so hopefully this will work!

sakoman, 2022-04-29:
@shr-project Well, I'm not sure how they got to the point of the qtbase build failing!  I get a failure on cmake-native.  I see this happening on dunfell with Ubuntu 22.04, so I suppose that is something I need to look into :-(

FWIW do_patch succeeds for both qtbase and qtdeclarative.

Will let you know if/when I make further progress

shr-project, 2022-04-30:
> Well, I'm not sure how they got to the point of the qtbase build failing! I get a failure on cmake-native. I see this happening on dunfell with Ubuntu 22.04, so I suppose that is something I need to look into :-( 

Are you using uninative? Then you have to cherry-pick https://git.openembedded.org/openembedded-core/commit/?id=e3da4da7e5da5bb9e1d360e2be2fdd5132e69320

libstdc++ was updated to version from gcc-12 build about a month ago in 22.04 which makes it incompatible with current uninative, the same works for gatesgarth just needs more uninative upgrades.

> FWIW do_patch succeeds for both qtbase and qtdeclarative.

That's good enough, will move it to gatesgarth later.

shr-project, 2022-04-30:
Looks like it was more backports than uninative I was using in oe-core gatesgarth I've already forgot about, all top 12 commits (from "m4: Do not use SIGSTKSZ") in:
https://git.openembedded.org/openembedded-core-contrib/log/?h=jansa/gatesgarth
I was using to build gatesgarth with 22.04 ubuntu and cmake-native still builds fine there.

sakoman, 2022-05-02:
They are using a dunfell/gatesgarth mix.  So it appears that only the uninative version bump is required for dunfell oe-core.  I'll include that in my next dunfell patch set.

shr-project, 2022-05-02:
Yes, for dunfell with 22.04 I had only boost-native which you've already merged (https://git.openembedded.org/openembedded-core-contrib/commit/?h=jansa/dunfell&id=64ba0d40a4c77a23778c51511f2d167e2056eea3) and the uninative 3.6 (https://git.openembedded.org/openembedded-core-contrib/commit/?h=jansa/dunfell&id=3da94094b11ea65fd714f0bf6763df84ed7603ac)