meta-qt5/meta-qt5 #489: Qt 5.15.7 LGPL

fmntf, 2022-11-23:
Qt has released the LGPL version of the framework [5.15.7](https://code.qt.io/cgit/qt/qtbase.git/tag/?h=v5.15.7-lts-lgpl)
Is there a plan to update this meta layer to such version? If not, are pull requests welcome?

Thank you!
Francesco

shr-project, 2022-11-23:
I don't use meta-qt5 anymore, but would accept PR.

h5b, 2023-01-10:
Hi  @fmntf, @shr-project ,

thanks for the update.

Following https://github.com/meta-qt5/meta-qt5/commit/2df0f760c200c932cfbb61a0ad684896dfe49e07 the layer now is based on Qt 5.15.7 LGPL / LTS release which is suitable for non-commercial use. Though, recipes for qtpdf and qtwebengine refer to version 5.15.10 which is not a LGPL release yet.

Is this perception correct or are these recipes missing only a version bump of ${PV}.

Kind regards,

Sebastian

h5b, 2023-01-10:
> Moreover, the Qt company released 5.15.8 as LGPL.
[...]

Thanks for your reply and yes i missed to mention that 5.15.8 has been released.

Though, i'm still puzzled if current versions are sane when it comes to non-commercial / OSS licensing:

qtpdf 5.15.10
qtwebengine 5.15.10

fmntf, 2023-01-10:
I'm not a lawyer, those versions look OK for me (and even for commercial purposes, if the LGPL3 is respected).
Even if the tag 5.15.10 has no "-lgpl" suffix the code is still licensed under the LGPL.

h5b, 2023-01-11:
@fmntf Thanks a lot!

closed 2024-01-09 by fmntf