meta-qt5/meta-qt5 #493: Project ERROR: Unknown module(s) in QT: multimedia

jahanzaib1234, 2022-12-11:
hi,

i want to use PyQt5.QtMultimedia but module not found in target machine to resolve this issue i check files of meta-qt5 in /meta-qt5/recipes-python/pyqt5 i found python3-pyqt5_5.15.1.bb file in .bb file i add QtMultimedia in PYQT_MODULES as bellow.

`PYQT_MODULES = "QtCore QtGui QtNetwork QtXml QtNetwork QtMultimedia QtQml ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'QtQuick QtWidgets QtQuickWidgets  QtMultimediaWidgets', '', d)}"`

QtCore and other modules are working but when i add QtMultimedia i get this error.

> python3-pyqt5/5.15.1-r0/PyQt5-5.15.1/QtMultimedia/QtMultimedia.pro
| Project ERROR: Unknown module(s) in QT: multimedia

jahanzaib1234, 2022-12-11:
im am using Respberry pi4 as a machine and using dunfell

jahanzaib1234, 2022-12-12:
kindly anyone look in this issue i am searching almost from last 10 days but i get nothing i am very thankful to anyone who helps.

If this thing is to hard to solve or no one want to answer this question kindly leave a comment about that because i am stuck at that point and now i just waiting to get answer from here kindly anyone response on this question.

jahanzaib1234, 2022-12-12:
@kergoth @denix0 @otavio @coldnew

jahanzaib1234, 2022-12-12:
@kraj Khem Raj

kraj, 2022-12-12:
We have upgraded to python 3.11 and older version of sip does not work with this, which pyqt modules need. We have added newer version of sip recipe to meta-oe but the python recipes in meta-qt5 will need to be ported. Perhaps using kirkstone LTS might be a good path for now for you.

jahanzaib1234, 2022-12-12:
@kraj thank you for your response and for showing path.

jahanzaib1234, 2022-12-12:
@kraj   I am new in yocto and embedded if you guide me litle bit more about krikstone LTS its helpful for me.

jahanzaib1234, 2023-01-07:
i resolve this issue by adding  qtmultimedia in DEPENDS and in RDEPENDS_${PN} like below.

`DEPENDS = "qtbase qtdeclarative qtquickcontrols2 qtmultimedia"
RDEPENDS_${PN} = "qtbase qtdeclarative qtquickcontrols2 qtquickcontrols2-mkspecs qtmultimedia"`

closed 2023-01-07 by jahanzaib1234