meta-qt5/meta-qt5 #495: [gatesgarth] QT build is broken

jviguera, 2022-12-12:
Hi all,

Seems that changes in the upstream QT repositories have broken **meta-qt5** for Yocto 3.2 Gatesgarth. Upstream branches **5.15.2** have been deleted and thus building in yocto fails at fetching with:

> ERROR: qtwayland-5.15.2+gitAUTOINC+3cc17177b1-r0 do_fetch: Fetcher failure: Unable to find revision 3cc17177b1b03053276eb6236fda137c588261a7 in branch 5.15.2 even from upstream
> ERROR: qtwayland-5.15.2+gitAUTOINC+3cc17177b1-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtwayland.git;name=qtwayland;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.
> ERROR: Logfile of failure stored in: ../ccimx6qpsbc/tmp/work/cortexa9t2hf-neon-mx6qdl-dey-linux-gnueabi/qtwayland/5.15.2+gitAUTOINC+3cc17177b1-r0/temp/log.do_fetch.4092915
> NOTE: recipe qtwayland-5.15.2+gitAUTOINC+3cc17177b1-r0: task do_fetch: Failed
> ERROR: Task (/mnt/ssd/jenkins/workspace/dey-github-gatesgarth/dey.gatesgarth/sources/meta-qt5/recipes-qt/qt5/qtwayland_git.bb:do_fetch) failed with exit code '1'
> 
> ERROR: qtquickcontrols-5.15.2+gitAUTOINC+7c29283041-r0 do_fetch: Fetcher failure: Unable to find revision 7c29283041a50def3ceca7de4471f211c9b30d4e in branch 5.15.2 even from upstream
> ERROR: qtquickcontrols-5.15.2+gitAUTOINC+7c29283041-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtquickcontrols.git;name=qtquickcontrols;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.
> ERROR: Logfile of failure stored in: ../ccimx6qpsbc/tmp/work/cortexa9t2hf-neon-dey-linux-gnueabi/qtquickcontrols/5.15.2+gitAUTOINC+7c29283041-r0/temp/log.do_fetch.4126828
> NOTE: recipe qtquickcontrols-5.15.2+gitAUTOINC+7c29283041-r0: task do_fetch: Failed
> ERROR: Task (/mnt/ssd/jenkins/workspace/dey-github-gatesgarth/dey.gatesgarth/sources/meta-qt5/recipes-qt/qt5/qtquickcontrols_git.bb:do_fetch) failed with exit code '1'
> 
> ERROR: qtdeclarative-native-5.15.2+gitAUTOINC+104eae5b17-r0 do_fetch: Fetcher failure: Unable to find revision 104eae5b17b0ec700391e9539ee3a4f638588194 in branch 5.15.2 even from upstream
> ERROR: qtdeclarative-native-5.15.2+gitAUTOINC+104eae5b17-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtdeclarative.git;name=qtdeclarative;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.
> ERROR: Logfile of failure stored in: ../ccimx6qpsbc/tmp/work/x86_64-linux/qtdeclarative-native/5.15.2+gitAUTOINC+104eae5b17-r0/temp/log.do_fetch.3859878
> NOTE: recipe qtdeclarative-native-5.15.2+gitAUTOINC+104eae5b17-r0: task do_fetch: Failed
> ERROR: Task (virtual:native:/mnt/ssd/jenkins/workspace/dey-github-gatesgarth/dey.gatesgarth/sources/meta-qt5/recipes-qt/qt5/qtdeclarative_git.bb:do_fetch) failed with exit code '1'
> 
> ERROR: qtdeclarative-5.15.2+gitAUTOINC+104eae5b17-r0 do_fetch: Fetcher failure: Unable to find revision 104eae5b17b0ec700391e9539ee3a4f638588194 in branch 5.15.2 even from upstream
> ERROR: qtdeclarative-5.15.2+gitAUTOINC+104eae5b17-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtdeclarative.git;name=qtdeclarative;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.
> ERROR: Logfile of failure stored in: ../ccimx6qpsbc/tmp/work/cortexa9t2hf-neon-dey-linux-gnueabi/qtdeclarative/5.15.2+gitAUTOINC+104eae5b17-r0/temp/log.do_fetch.3872350
> NOTE: recipe qtdeclarative-5.15.2+gitAUTOINC+104eae5b17-r0: task do_fetch: Failed
> ERROR: Task (/mnt/ssd/jenkins/workspace/dey-github-gatesgarth/dey.gatesgarth/sources/meta-qt5/recipes-qt/qt5/qtdeclarative_git.bb:do_fetch) failed with exit code '1'
> 
> ERROR: qtbase-native-5.15.2+gitAUTOINC+40143c189b-r0 do_fetch: Fetcher failure: Unable to find revision 40143c189b7c1bf3c2058b77d00ea5c4e3be8b28 in branch 5.15.2 even from upstream
> ERROR: qtbase-native-5.15.2+gitAUTOINC+40143c189b-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtbase.git;name=qtbase;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.
> ERROR: Logfile of failure stored in: ../ccimx6qpsbc/tmp/work/x86_64-linux/qtbase-native/5.15.2+gitAUTOINC+40143c189b-r0/temp/log.do_fetch.3810280
> NOTE: recipe qtbase-native-5.15.2+gitAUTOINC+40143c189b-r0: task do_fetch: Failed
> ERROR: Task (/mnt/ssd/jenkins/workspace/dey-github-gatesgarth/dey.gatesgarth/sources/meta-qt5/recipes-qt/qt5/qtbase-native_git.bb:do_fetch) failed with exit code '1'
> 
> ERROR: qtbase-5.15.2+gitAUTOINC+40143c189b-r0 do_fetch: Fetcher failure: Unable to find revision 40143c189b7c1bf3c2058b77d00ea5c4e3be8b28 in branch 5.15.2 even from upstream
> ERROR: qtbase-5.15.2+gitAUTOINC+40143c189b-r0 do_fetch: Fetcher failure for URL: 'git://code.qt.io/qt/qtbase.git;name=qtbase;branch=5.15.2;protocol=git'. Unable to fetch URL from any source.

Not sure what is the proper fix for this. Maybe changing the not-existing branch for **nobranch=1** in the recipes is the way to go, but I'm not sure if this has side effects.

shr-project, 2022-12-12:
I'm aware of this issue, but I don't plan to add nobranch in meta-qt5, see https://github.com/meta-qt5/meta-qt5/pull/377#pullrequestreview-585699882

RogueWarrior65, 2023-01-02:
I'm running into this problem as well.  Is there a workaround for it?

madscientist42, 2023-01-20:
Not really.  The real answer is to quit picking "Z" branches where it's versioned X.Y.Z or use a specific githash for the commit in question.  Messy, but the problem people have is that there's upstream devs that **_absolutely don't have a handle on this._**- they will delete the offending tags or worse.  It was rather annoying trying to use the version tags that FreeRTOS has because they put the tags in and apparently deleted them or propagated them from a mirror improperly.

@shr-project : SOMETHING needs to be done.  It makes it where only Master is safe for people to use- and at some point, we can't do reproducible builds because your stuff and the Qt projects' is nothing to rely upon.  If I'm mirroring it because I can't rely on it's history, etc. to be consistent, why bother?

shr-project, 2023-01-20:
SRCREV uses specific git hashes, but bitbake fetcher also checks that these git hashes are on the specified branch. Which isn't true after they deleted the X.Y.Z branches which they asked me to use instead of X.Y branches where the releases used to be downmerged (a I was waiting with meta-qt5 updates until that happened).

Now they disable this check everywhere with nobranch=1 which is stupid, but I'm not The Qt Company employee, I was never paid by them and I've done enough to get their shit together (see https://bugreports.qt.io/browse/QTQAINFRA-3091 and https://github.com/meta-qt5/meta-qt5/issues/349#issuecomment-691077189 ).

Merging nobranch=1 in meta-qt5 active branches wouldn't fix projects which are using older revision (for whatever reason, I know about big project still using 5.12.11 instead of warrior branch head with 5.12.12 and they have the same issue, for them I've created jansa/warrior-overrides branch where this is fixed (with only one nobranch in qtbase recipe, because only qtbase v5.12.12 tag isn't in any branch, other were downmerged to 5.12).

Others need to deal with it on their own or ask TQC to restore the branches as I said there "Branches are cheap, breaking existing projects like meta-qt5 without good reason is not.".