meta-qt5/meta-qt5 #516: Update to Qt 5.15.9 LGPL (opensource)

bernhardHartleb, 2023-04-09:
Release announcement here:
[https://lists.qt-project.org/pipermail/announce/2023-April/000406.html](https://lists.qt-project.org/pipermail/announce/2023-April/000406.html)

I am willing to make a PR if desired.
Thank you.

bernhardHartleb, 2023-04-09:
For reference:
[https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/5.15.8/release-note.md](https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/5.15.8/release-note.md)
[https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/5.15.9/release-note.md](https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/5.15.9/release-note.md)

shr-project, 2023-04-09:
> I am willing to make a PR if desired.

Please go ahead.

chakuzo, 2023-04-11:
I am also interested, also support if needed.

fmntf, 2024-01-09:
This issue should be closed since 5.15.9 is already on master branch.

closed 2024-01-09 by otavio