meta-qt5/meta-qt5 #518: SRCREV_FORMAT with underscores generates Debian packages with invalid versions

czw, 2023-04-14:
Some of the recipes, for example `qtlocation`, contains version numbers with underscores in them. The packages will build just fine, but adding the resulting `qtlocation_5.15.3+git0+9909253b0f_d3101bbc22-r0_amd64.deb` file to a Debian package repository with `reprepro` yields the following error:

    Character '_' not allowed in version: '5.15.3+git0+9909253b0f_d3101bbc22-r0'

I am willing to make a PR if desired. All I need to know is if it's OK to replace the underscore with a plus sign.

shr-project, 2023-04-14:
Every SRCREV_FORMAT I've found is using underscore as separator, so this would be better to discuss on oe-core ML first as it doesn't affect just qt5 recipes.

czw, 2023-04-16:
Correct, this is probably best done in the deb archive creation side. Closing.

closed 2023-04-16 by czw