meta-qt5/meta-qt5 #522: Most recent commit adding assimp is broken

reatmon-ti, 2023-04-22:
The most recent commit: 
https://github.com/meta-qt5/meta-qt5/commit/962f45e2661295bab10e17aacf7914ffe8ac59ae

Is broken.  The assimp repo does not contain a branch named assimp_5.0_release.

shr-project, 2023-04-27:
They deleted the branch recently, it doesn't fail for me as I still have it in my local DL_DIR and PREMIRROR:

```
downloads/git2/github.com.assimp.assimp.git $ git branch -a --contains 8f0c6b04b2257a520aaab38421b2e090204b69df
  assimp_5.0_release
```

```
downloads/git2/github.com.assimp.assimp.git $ git remote prune origin
Pruning origin
URL: https://github.com/assimp/assimp.git
 * [pruned] refs/heads/acgessler-patch-4
 * [pruned] refs/heads/assimp_5.0_release
 * [pruned] refs/heads/coverity_scan
 * [pruned] refs/heads/custom_allocator
 * [pruned] refs/heads/feature/configured_precision
 * [pruned] refs/heads/feature/opengex_missing_features
 * [pruned] refs/heads/fix_android
 * [pruned] refs/heads/fix_models
.....
```

it's also possible to use qtquick3d-assimp which still has this revision:
```
downloads/git2/code.qt.io.qt.qtquick3d-assimp.git $ git branch -a --contains 8f0c6b04b2257a520aaab38421b2e090204b69df
  qt6_assimp
```

Can you send a patch to use qtquick3d-assimp ?

reatmon-ti, 2023-04-28:
Ok. Working on creating and testing the patch today.  I'll submit two pull requests for both kirkstone and master since we are seeing it on both branches.

reatmon-ti, 2023-04-28:
Ok.  Pull requests for master and kirkstone submitted.