meta-qt5/meta-qt5 #552: ERROR: qtwebkit-5.15.12+git-r0 do_compile

leimaohui, 2024-01-23:
With the last poky, qtwebkit is failed. The log is  as the following: 
```
NOTE: Executing Tasks
ERROR: qtwebkit-5.15.12+git-r0 do_compile: ExecutionError('/ubinux-dev/ubinux001/build-ubinux/ubinux-202402/build-yocto/tmp/work/cortexa57-poky-linux/qtwebkit/5.15.12+git/temp/run.do_compile.2479616', 1, None, None)
ERROR: Logfile of failure stored in: /ubinux-dev/ubinux001/build-ubinux/ubinux-202402/build-yocto/tmp/work/cortexa57-poky-linux/qtwebkit/5.15.12+git/temp/log.do_compile.2479616
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /ubinux-dev/ubinux001/build-ubinux/ubinux-202402/build-yocto/tmp/work/cortexa57-poky-linux/qtwebkit/5.15.12+git/git --target all --
| Error: could not load cache
| WARNING: exit code 1 from a shell command.
ERROR: Task (/ubinux-dev/common/home/ubinux/community/meta-qt5/recipes-qt/qt5/qtwebkit_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3143 tasks of which 995 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
```

The commitID of poky and meta-qt5  is the following:
```
poky: 1df683df5c2173c3f38cfd24f40a731a4373c88d
meta-qt5: 526b61e10908f733701c2bf0e4db629b428c0384
```

And I found that the following submit of poky caused this issue, but I don't know why.
```
commit c665a2c93337fc372207af604c21bdea5babb4e5
Author: Richard Purdie <richard.purdie@linuxfoundation.org>
Date:   Tue Jan 9 13:47:36 2024 +0000

    bitbake: ast: Fix EXPORT_FUNCTIONS bug

    If you have two classes, both of which set EXPORT_FUNCTIONS for the same funciton
    and a standard funciton definition for the function that is exported, the export
    function can sometimes overwrite the standard one.

    The issue is that the internal flag the code uses isn't ovweritten if the variable
    is giving a new value. Fix the issue by using a comment in the code that is injected
    so that we know if it is ours or not.

    Also add some testing for EXPORT_FUNCTIONS, not perfect but a start.

    (Bitbake rev: 66306d5151acb0a26a171c338d8f60eb9eb16c6b)

    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>

 bitbake/lib/bb/parse/ast.py   |  9 +++++----
 bitbake/lib/bb/tests/parse.py | 98 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 2 files changed, 103 insertions(+), 4 deletions(-)
```
I wonder if there are anybody know how to fix this issue.

atvcaptain, 2024-01-23:
yes i have same issue with webkit , i use master branch

denix0, 2024-01-23:
The PR with the fix has been open for several days now:
https://github.com/meta-qt5/meta-qt5/pull/551

@shr-project would you mind taking a look at the fix, please? Thanks!

denix0, 2024-01-23:
Cc: @otavio

leimaohui, 2024-01-24:
> The PR with the fix has been open for several days now: #551
> 
> @shr-project would you mind taking a look at the fix, please? Thanks!

@denix0 
It works. Thank you.

closed 2024-01-24 by leimaohui