meta-qt5/meta-qt5 #567: "layer: Reinstate support for older OE releases" throws error

htot, 2024-05-21:
Since support for older OE releases has been reinstated bitbake throws:
...
" which triggered exception AttributeError: 'NoneType' object has no attribute 'split'"
    
Caused by ([here](https://github.com/meta-qt5/meta-qt5/blob/d8eeef0bfd84672c7919cd346f25f7c9a98ddaea/conf/layer.conf#L44)):
`"${@'scarthgap' if 'scarthgap' in d.getVar('LAYERSERIES_CORENAMES').split() else 'legacy'}/*/*/*.bbappend"`
    
This can be fixed by:
`"${@bb.utils.contains('LAYERSERIES_CORENAMES', 'scarthgap', 'scarthgap', 'legacy', d)}/*/*/*.bbappend"`
which does the same.

shr-project, 2024-05-22:
can you share output from:
bitbake-getvar LAYERSERIES_CORENAMES
in your setup? it's strange that it wouldn't be set.

htot, 2024-05-22:
```
ferry@delfion:~/tmp/edison-intel/my/edison-morty/out/linux64/build$ bitbake-getvar LAYERSERIES_CORENAMES
NOTE: Starting bitbake server...
#
# $LAYERSERIES_CORENAMES [3 operations]
#   set /home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta/conf/layer.conf:10
#     "scarthgap"
#   del cookerdata.py:414 [parseConfigurationFiles]
#     ""
#   set cookerdata.py:455 [parseConfigurationFiles]
#     "scarthgap"
# pre-expansion value:
#   "scarthgap"
LAYERSERIES_CORENAMES="scarthgap"
```

shr-project, 2024-05-22:
Strange, they why would it be NoneType? I'm not seeing this issue in my local builds and @marex confirmed it works for him as well (in other layers as well).

htot, 2024-05-23:
Could it be a python bug? I now just updated from Ubuntu 23.10 to 24.04 and python is now changed from 3.11 to 3.12.

Now, when I add the line:
```
FERRY = "${@'scarthgap' if 'scarthgap' in d.getVar('LAYERSERIES_CORENAMES').split else 'legacy'}/*/*/*.bbappend"
```
and type `bitbake-getvar FERRY` I get:
```
...
bb.data_smart.ExpansionError: Failure expanding variable FERRY, expression was ${@'scarthgap' if 'scarthgap' in d.getVar('LAYERSERIES_CORENAMES').split else 'legacy'}/*/*/*.bbappend which triggered exception TypeError: argument of type 'builtin_function_or_method' is not iterable
```

when I remove `.split` I get:
```
ferry@delfion:~/tmp/edison-intel/my/edison-morty/out/linux64/build$ bitbake-getvar FERRY
NOTE: Starting bitbake server...
#
# $FERRY
#   set /home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-qt5/conf/layer.conf:44
#     "${@'scarthgap' if 'scarthgap' in d.getVar('LAYERSERIES_CORENAMES') else 'legacy'}/*/*/*.bbappend"
FERRY="scarthgap/*/*/*.bbappend"
```

shr-project, 2024-05-23:
do you have:

```d.getVar('LAYERSERIES_CORENAMES').split```

or

```d.getVar('LAYERSERIES_CORENAMES').split()```

in FERRY?

htot, 2024-05-24:
Oops, I had `d.getVar('LAYERSERIES_CORENAMES').split`

And on a Ubuntu 22.04.4 / kirkstone setup `d.getVar('LAYERSERIES_CORENAMES').split()` seems to be working well.

Let me try on the 24.04 / scarthgap setup tonight.

htot, 2024-05-24:
But still (full error message, manual LF's inserted):
```
ferry@delfion:~/tmp/edison-intel/my/edison-morty/out/linux64/build$ bitbake -k edison-image
ERROR: Unable to parse Var <BBFILES>
Traceback (most recent call last):
  File "Var <BBFILES>", line 1, in <module>
bb.data_smart.ExpansionError: Failure expanding variable BBFILES, expression was  
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta/recipes-*/*/*.bb 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-poky/recipes-*/*/*.bb             
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-poky/recipes-*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-yocto-bsp/recipes-*/*/*.bb             
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-yocto-bsp/recipes-*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-oe/recipes-*/*/*.bb 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-oe/recipes-*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-python/recipes*/*/*.bb 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-python/recipes*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-networking/recipes-*/*/*.bb             
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-openembedded/meta-networking/recipes-*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-qt5/recipes-*/*/*.bb 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-qt5/recipes-*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-qt5/compat/${@'scarthgap' if 'scarthgap' in d.getVar('LAYERSERIES_CORENAMES').split() else 'legacy'}/*/*/*.bbappend 
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-intel/recipes-*/*/*.bb             
/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta-intel/recipes-*/*/*.bbappend 
which triggered exception AttributeError: 'NoneType' object has no attribute 'split'
The variable dependency chain for the failure is: BBFILES -> BBFILES[:=]
```
So, while the expression parses by itself, it won't while appended to BBFILES.

Surprising, I would have expected it to be expanded, then appended, but it seems the other way around.

shr-project, 2024-05-24:
Does it happen if you include only meta-qt5 and oe-core in BBLAYERS?

htot, 2024-05-24:
I don't have `oe-core`, I have (paths shortned)
```
BBLAYERS ?= " \
  .../out/linux64/poky/meta \
  .../out/linux64/poky/meta-poky \
  .../out/linux64/poky/meta-yocto-bsp \
  .../out/linux64/poky/meta-openembedded/meta-oe \
  .../out/linux64/poky/meta-openembedded/meta-python \
  .../out/linux64/poky/meta-openembedded/meta-networking \
  .../out/linux64/poky/meta-qt5 \
  .../out/linux64/poky/meta-intel \
  .../meta-intel-edison/meta-intel-edison-bsp \
  .../meta-intel-edison/meta-intel-edison-distro \
  .../meta-acpi \
  \
  "
```

But I do note this, my fix (bb.utils.contains()) also doesn't work. It resolves to legacy. So, that can only mean that at the time of expansion `LAYERSERIES_CORENAMES` is not yet set. That also explains the error in split().

shr-project, 2024-05-24:
poky/meta is oe-core

It could be one of the other layers you have enabled which is causing BBFILES to be parsed sooner or something, knowing which one might explain why we're not seeing the failure you're seeing.

htot, 2024-05-24:
In that case I get `ERROR: Layer 'qt5-layer' depends on layer 'openembedded-layer', but this layer is not enabled in your configuration`

shr-project, 2024-05-24:
ok, then add meta-oe as well

htot, 2024-05-24:
Found it. Those layers had
```
BBFILES := "${BBFILES} ${LAYERDIR}/recipes-*/*/*.bb \
	${LAYERDIR}/recipes-*/*/*.bbappend"
```

after changing to

```
BBFILES += "${LAYERDIR}/recipes-*/*/*.bb \
            ${LAYERDIR}/recipes-*/*/*.bbappend"
```

it works. That was code from 2015 :-)

You've been a great help, thanks. Closing.

closed 2024-05-24 by htot

shr-project, 2024-05-24:
Which of those layers?

htot, 2024-05-24:
Both mine `meta-intel-edison-bsp` and `meta-intel-edison-distro` (see edited above).