meta-qt5/meta-qt5 #573: do_compile error with gcc14

leimaohui, 2024-06-20:
After poky upgrade gcc to 14, qtwebkit build failed.
----------------------------------------------------------------
/home/leimh/build-ubinux-daily/tmp/work/aarch64-ubinux-linux/qtwebkit/5.15.13+git/recipe-sysroot/usr/include/unicode/localpointer.h:561:26: error: parameter declared 'auto'
  561 | template <typename Type, auto closeFunction>
      |                          ^~~~
/home/leimh/build-ubinux-daily/tmp/work/aarch64-ubinux-linux/qtwebkit/5.15.13+git/recipe-sysroot/usr/include/unicode/localpointer.h:573:76: error: template argument 2 is invalid
  573 |     explicit LocalOpenPointer(std::unique_ptr<Type, decltype(closeFunction)> &&p)
      |                                                                            ^
/home/leimh/build-ubinux-daily/tmp/work/aarch64-ubinux-linux/qtwebkit/5.15.13+git/recipe-sysroot/usr/include/unicode/localpointer.h:583:78: error: template argument 2 is invalid
  583 |     LocalOpenPointer &operator=(std::unique_ptr<Type, decltype(closeFunction)> &&p) {
      |                                                                              ^
/home/leimh/build-ubinux-daily/tmp/work/aarch64-ubinux-linux/qtwebkit/5.15.13+git/recipe-sysroot/usr/include/unicode/localpointer.h:599:59: error: template argument 2 is invalid
  599 |     operator std::unique_ptr<Type, decltype(closeFunction)> () && {

shr-project, 2024-06-20:
This one isn't cause by gcc-14 but icu upgrade I believe. There is another gcc-14 related error I've mentioned in https://github.com/meta-qt5/meta-qt5/pull/570. Patches welcome from someone still using this (I don't).

@reatmon-ti you've repoted the same over e-mail, did you fix it already?

reatmon-ti, 2024-06-20:
No.  I found that it was not related to ICU as well like I initially 
thought.

We looked further on our end and found that we had recently removed some 
internal tools that were using qtwebkit.  So our final solution was to 
just remove all of the dependencies on qtwebkit and stop using it all.

shr-project, 2025-04-22:
I think I've fxed this in https://github.com/meta-qt5/meta-qt5/pull/604 but the same is reproducible with qtlocation and geoservices_mapboxgl PACKAGECONFIG enabled.

closed 2025-04-22 by shr-project