meta-qt5/meta-qt5 #597: master: qtbase do_install failed after bump to LTS

hiagofranco, 2025-03-17:
Hi everyone, after https://github.com/meta-qt5/meta-qt5/pull/593 was merged, now we are seeing this do_install failure into our master builds:

```
sed: can't read /workdir/oe/tmp/work/aarch64-oe-linux/qtbase/5.15.16+git/image/usr/share/examples/widgets/tools/plugandpaint/plugins/libpnp_basictools.prl: No such file or directory
```

Looks like commit ee7907bbdf7397571d3af4ec8627344a84fdf495 (“qtbase: Replace reference to build directory in library dependency file“) is causing this issue. Can you please help us to fix this issue? Should we revert this commit?

Thanks

igifalk, 2025-03-18:
Take a look into issue #596. There is a patch which fix the message.
Currently i have no master build. The patch is made in branch scarthgap but should work in master too.

hiagofranco, 2025-03-18:
Hi @igifalk, thanks for sharing. As this is a duplicate of #596, I will close this issue.

closed 2025-03-18 by hiagofranco