meta-qt5/meta-qt5 #619: qtwebengine: build ERROR with gcc-15

titanobi, 2025-07-12:
[12657/23595] CXX obj/third_party/blink/public/mojom/service_worker/storage/navigation_preload_state.mojom.o
[12658/23595] CXX obj/third_party/blink/public/mojom/tokens/tokens/tokens.mojom-test-utils.o
[12659/23595] CXX obj/third_party/blink/public/mojom/service_worker/storage/service_worker_registration_options.mojom.o
[12660/23595] CXX obj/third_party/blink/public/mojom/tokens/tokens/tokens.mojom.o
[12661/23595] CXX obj/third_party/blink/public/mojom/usb/usb/web_usb_service.mojom-test-utils.o
[12662/23595] STAMP obj/third_party/devtools-frontend/src/front_end/client_variations/bundle.stamp
[12663/23595] STAMP obj/third_party/devtools-frontend/src/front_end/color_picker/bundle.stamp
[12664/23595] STAMP obj/third_party/devtools-frontend/src/front_end/component_helpers/bundle.stamp
[12665/23595] STAMP obj/third_party/devtools-frontend/src/front_end/cookie_table/bundle.stamp
[12666/23595] STAMP obj/third_party/devtools-frontend/src/front_end/coverage/bundle.stamp
[12667/23595] STAMP obj/third_party/devtools-frontend/src/front_end/emulation/bundle.stamp
[12668/23595] STAMP obj/third_party/devtools-frontend/src/front_end/event_listeners/bundle.stamp
[12669/23595] STAMP obj/third_party/devtools-frontend/src/front_end/extensions/bundle.stamp
[12670/23595] STAMP obj/third_party/devtools-frontend/src/front_end/har_importer/bundle.stamp
[12671/23595] STAMP obj/third_party/devtools-frontend/src/front_end/inline_editor/bundle.stamp
[12672/23595] STAMP obj/third_party/devtools-frontend/src/front_end/layer_viewer/bundle.stamp
[12673/23595] STAMP obj/third_party/devtools-frontend/src/front_end/profiler/bundle.stamp
[12674/23595] STAMP obj/third_party/devtools-frontend/src/front_end/search/bundle.stamp
[12675/23595] STAMP obj/third_party/devtools-frontend/src/front_end/snippets/bundle.stamp
[12676/23595] STAMP obj/third_party/devtools-frontend/src/front_end/third_party/acorn-logical-assignment/acorn-logical-assignment-copy-javascript.stamp
[12677/23595] STAMP obj/third_party/devtools-frontend/src/front_end/third_party/acorn-numeric-separator/acorn-numeric-separator-copy-javascript.stamp
[12678/23595] STAMP obj/third_party/devtools-frontend/src/front_end/third_party/acorn/acorn-copy-javascript.stamp
[12679/23595] STAMP obj/third_party/devtools-frontend/src/front_end/third_party/marked/marked-copy-javascript.stamp
[12680/23595] STAMP obj/third_party/devtools-frontend/src/front_end/timeline_model/bundle.stamp
[12681/23595] STAMP obj/third_party/devtools-frontend/src/front_end/workspace_diff/bundle.stamp
[12682/23595] ACTION //third_party/devtools-frontend/src/inspector_overlay:build_inspector_overlay(/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/toolchain:target)
FAILED: resources/inspector_overlay/common.css resources/inspector_overlay/tool_distances.js resources/inspector_overlay/tool_highlight_grid.js resources/inspector_overlay/tool_highlight.js resources/inspector_overlay/tool_paused.js resources/inspector_overlay/tool_screenshot.js resources/inspector_overlay/tool_source_order.js resources/inspector_overlay/tool_viewport_size.js 
python3 ../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/scripts/build/build_inspector_overlay.py common.css 1200 tool_highlight.js 90000 tool_highlight_grid.js 70000 tool_distances.js 5000 tool_paused.js 10000 tool_screenshot.js 5000 tool_viewport_size.js 5000 tool_source_order.js 11000 --input_path gen/third_party/devtools-frontend/src/inspector_overlay --output_path resources/inspector_overlay --rollup_plugin ../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/inspector_overlay/loadCSS.rollup.js
Usage: ../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/scripts/build/build_inspector_overlay.py filename_1 max_size_1 filename_2 max_size_2 ... filename_N max_size_N --input_path <input_path> --output_path <output_path>
Traceback (most recent call last):
  File "/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/core/release/../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/scripts/build/build_inspector_overlay.py", line 107, in <module>
    sys.exit(main(sys.argv))
  File "/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/core/release/../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/scripts/build/build_inspector_overlay.py", line 92, in main
    rollup(input_path, output_path, filename, max_size,
  File "/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/core/release/../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/scripts/build/build_inspector_overlay.py", line 64, in rollup
    raise Exception("rollup failed: " + error)
Exception: rollup failed: [!] Error: Invalid --plugin argument format: "../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/inspector_overlay/loadCSS.rollup.js"
Error: Invalid --plugin argument format: "../../../../sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/inspector_overlay/loadCSS.rollup.js"
    at loadAndRegisterPlugin (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/shared/commandPlugins.js:413:19)
    at /home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/shared/commandPlugins.js:392:55
    at Array.forEach (<anonymous>)
    at Object.addCommandPluginsToInputOptions (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/shared/commandPlugins.js:392:35)
    at loadConfigFromCommand (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/bin/rollup:1655:20)
    at getConfigs (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/bin/rollup:2139:12)
    at runRollup (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/bin/rollup:2117:49)
    at Object.<anonymous> (/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/sources/qtwebengine-5.15.16+git/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules/rollup/dist/bin/rollup:2159:5)
    at Module._compile (node:internal/modules/cjs/loader:1364:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1422:10)

[12683/23595] CXX obj/third_party/blink/public/mojom/usb/usb/web_usb_service.mojom.o
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile.gn_run:367: run_ninja] Error 1
make[3]: Leaving directory '/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/core'
make[2]: *** [Makefile:82: sub-gn_run-pro-make_first] Error 2
make[2]: Leaving directory '/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src/core'
make[1]: *** [Makefile:79: sub-core-make_first] Error 2
make[1]: Leaving directory '/home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/build/src'
[... cut ...]
	#4: base_do_compile, /home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/temp/run.do_compile.2400510, line 161
	#5: do_compile, /home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/temp/run.do_compile.2400510, line 155
	#6: main, /home/obi/build-enviroment-5.6/builds/openaaf/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/qtwebengine/5.15.16+git/temp/run.do_compile.2400510, line 203

shr-project, 2025-07-12:
Looks the same as what I've seen in https://github.com/meta-qt5/meta-qt5/commit/e10fdabb88eb85d33e9483f44891710a95510c86 are you sure this other issue from python is related to gcc-15?

WXbet, 2025-07-13:
Try this patch in `src/3rdparty`:
```
diff --git a/chromium/third_party/devtools-frontend/src/inspector_overlay/BUILD.gn b/chromium/third_party/devtools-frontend/src/inspector_overlay/BUILD.gn
index 83041bcb8..2c1adcdf4 100644
--- a/chromium/third_party/devtools-frontend/src/inspector_overlay/BUILD.gn
+++ b/chromium/third_party/devtools-frontend/src/inspector_overlay/BUILD.gn
@@ -72,7 +72,7 @@ action("build_inspector_overlay") {
   ]
   if (use_rollup) {
     args += [
-      "--rollup_plugin",
+      "--bundleConfigAsCjs --rollup_plugin",
       rebase_path("loadCSS.rollup.js", root_build_dir),
     ]
   }
```

shr-project, 2025-07-13:
Please send a PR.

WXbet, 2025-07-13:
https://github.com/meta-qt5/meta-qt5/pull/620